NVIDIA-AI-IOT/jetbot #98: jetbot wireless with 4A barrel power

nathangeorge1, 2019-06-28:
Hello,

Is there a way to run the Nano with 4A using the barrel jack but also wireless? There are many ways to get this using a wall outlet, but I can't seem to find something using a battery. I believe it also needs around 5.35V.

My current thoughts are to plug the [recommended battery](https://www.amazon.com/gp/product/B07H6LB4J4/ref=as_li_ss_tl?ie=UTF8&linkCode=sl1&tag=jbstore074-20&linkId=28aad75b99de8dc39f7a8ee2fc5a15aa&language=en_US) into the barrel jack using a microUSB->DC connector, but will that supply the correct voltage?

Thanks,
Nathan

FPSychotic, 2019-06-29:
Maybe you can check the voltage with a polimeter before connect it. Probably someone close to you have one.
Otherwise you can contact to the battery vendor via amazon questions sections, Im sure they  will give you more precise info

nathangeorge1, 2019-07-02:
The recommended battery didn't work with the barrel jack, and I believe it's because the battery only outputs a max of 5V, but after some drops, it would really need around [5.35V](https://devtalk.nvidia.com/default/topic/1051057/jetson-nano/jetson-nano-barrel-jack-power-not-working/post/5356278/?offset=31#5357250).

Is there a recommended battery for the barrel jack, that could output 5.35V?

wujupi, 2019-08-22:
Recommended 12v lithium battery pack + 12V to 5V Power Converter LM2596S(Max current 3A) 

or XY-3606 (Max current 5A) 

https://www.amazon.com/YJYdada-6800-18000mAh-Rechargeable-Lithium-18000mAh/dp/B07G4171RY/ref=sr_1_4?keywords=12v+lithium+battery+pack&qid=1566482961&s=gateway&sr=8-4

https://www.amazon.com/BBOXIM-Step-Down-Voltage-Converter-XY-3606/dp/B07VT1S85X/ref=sr_1_1?keywords=XY-3606&qid=1566483027&s=gateway&sr=8-1

https://www.amazon.com/Cenrykay-Step-Down-Module-Converter-LM2596S/dp/B07H3YR9BG/ref=sr_1_1_sspa?keywords=lm2596s&qid=1566483052&s=gateway&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFLVElXWVg3SUwwTzcmZW5jcnlwdGVkSWQ9QTAyNjc0NjkzSE5BWDRFME5QSlJMJmVuY3J5cHRlZEFkSWQ9QTAxMzE3MzEzRFpRQ0dJWEY2R0EyJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

nathangeorge1, 2019-08-22:
@wujupi that's perfect; thanks!

closed 2019-08-22 by nathangeorge1