NVIDIA-AI-IOT/jetbot #100: Download of files from Jetbot to windows in Jupyter fails due to forbidden

dhjmackay, 2019-07-04:
Running the Jetbot through its examples but got hung up in collision detection when trying to download the dataset.zip file to my workstation (a Windows 10 desktop). Any file I try to download (by right clicking on the file name in Jupyter and selecting "Download") fails with a "Forbidden" message. I'm using Chrome and their advice for this message is to change the file's permissions on the server. The site is flagged by the browser as being "not secure" but I see no way to change that in the browser. There must be something on the JetBot side that is tripping the not secure flag. Any help would be greatly appreciated.

dvillevald, 2019-07-04:
Try Internet Explorer for downloading, it worked for me.

dhjmackay, 2019-07-04:
IE works. Seems that Chrome's Jupyter window doesn't prompt for a password on opening the Jetbot window while IE does (I was wondering why the instructions always had an extra password step that I wasn't seeing). The password clearly is necessary to secure the portal. Good catch though I'm kinda surprised I'm the only one to have noticed this.

closed 2019-07-04 by dhjmackay