NVIDIA-AI-IOT/jetbot #102: Nvidia Kaya vs Nvidia JetBot

unbug, 2019-07-08:
I saw the video of [Nvidia Kaya](https://docs.nvidia.com/isaac/isaac/apps/tutorials/doc/assemble_kaya.html) on [YouTube](https://www.youtube.com/watch?v=5c9hJHmMKAs), it's pretty cool, but I didn't find it on GitHub, so what's the difference between both of them?
[image]

kenvidia, 2019-07-08:
Hi @unbug ,

You can find Kaya available for download on GitHub here: https://github.com/nvidia-isaac/kaya-robot 

Kaya was created as a demonstration and development platform for [NVIDIA's Isaac SDK](https://developer.nvidia.com/isaac-sdk) for robotics, running on the Jetson Nano. Kaya is capable of holonomic motion (ie. moving in any direction and rotating on the spot), and has wheel encoders and an IMU which provide precise odometry data. It is also equipped with a very sophisticated sensor in the Intel Realsense D435, which is a depth camera using active IR. This allows for mapping and localisation, object avoidance, etc.

We are also planning to release a version of Kaya which uses an inexpensive Lidar and a CSI camera in a few months. In the meantime, you can find sample apps for the Realsense-based Kaya included in the [NVIDIA Isaac SDK](https://developer.nvidia.com/isaac/downloads). Here is a link to the [Isaac SDK documentation](https://docs.nvidia.com/isaac/isaac/doc/index.html). Documentation for Kaya assembly has also been copied over to [GitHub](https://github.com/nvidia-isaac/kaya-robot/blob/master/doc/assemble_kaya.md).

dax-think, 2019-09-28:
Hi @kenvidia 

May I know when the new version of Kaya with Lidar and CSI camera will be released?

Thanks!

closed 2019-09-28 by unbug

kenvidia, 2019-09-30:
Hi @dax-think ,

I'm not sure exactly when we'll be releasing this, but I expect it will be packaged with some other NVIDIA Isaac release items the end of the year.