NVIDIA-AI-IOT/jetbot #105: remote desktop

seba-eng, 2019-07-11:
Is it possible to remote desktop (from windows 10) into the jetson as an easy solution to run some GUI programs? I installed xrdp on the jetson, but I can not get further than the login screen. The session always just terminates after that. Is the jetbot account maybe automatically logged in after power up?

jaybdub, 2019-07-16:
Hi seba-eng,

Thanks for reaching out!  I'm not sure if it's the best solution, I've used X-forwarding before to render some simple GUI programs with Jetson.

To answer your second question, yes the JetBot automatically logs in at boot.

Please let me know if you have any other questions.

Best,
John

wujupi, 2019-08-22:
Please use Vino-server + dummy plug

install vino
`sudo apt-get install vino`

Enabling Desktop Sharing
https://blog.hackster.io/getting-started-with-the-nvidia-jetson-nano-developer-kit-43aa7c298797

https://www.amazon.com/Headless-Display-Emulator-Generation-Single/dp/B07FB8GJ1Z/ref=sr_1_1_sspa?keywords=dummy+plug&qid=1566481551&s=gateway&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyOTM2MzgzOVlDS0g4JmVuY3J5cHRlZElkPUEwNzU1Njg0WVhDWE1DSUVXV0o0JmVuY3J5cHRlZEFkSWQ9QTAyMjQ0NjMzT0hUVEc1WkZJUVE2JndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

closed 2022-07-15 by jaybdub