NVIDIA-AI-IOT/jetbot #106: PiOLED display out of stock - alternatives?

vnguyen13, 2019-07-12:
Hi there,

The PiOLED display appears to be out of stock everywhere. Is there an alternative that could possibly work with this?

DouglasPearless, 2019-07-12:
If you are prepared to wait a week, I have been working with another and have completely redesigned the chassis so it takes a much bigger battery and a cheaper OLED display and a USB-C PD board to power it all...

It will be published on GitHub as a fork off the original repo.

Cheers
Douglas

vnguyen13, 2019-07-12:
That’s awesome! By any chance, is it going to be using the same Jetbot image? Because I’ve printed nearly 15 of the bodies so far for running a class... only to see that the pioled header is out of stock.

DouglasPearless, 2019-07-13:
I also have a design that allows a cheaper OLED to be mounted on a bracket that screws in using the existing mounting holes if you want that uploaded to?

vnguyen13, 2019-07-13:
Interesting! I’m curious to see what that looks like, sure!

DouglasPearless, 2019-07-14:
Have a look here https://github.com/DouglasPearless/jetbot for my updates …

vnguyen13, 2019-07-15:
Thanks Douglas! Is the image for the jetbot the same? As in, can we hotswap the new LCD screen with the one you have?

AustinPavlas, 2019-07-16:
Would it be possible to use this display instead of the PiOLED one? https://www.adafruit.com/product/931

It uses the same driver and I have been able to manually make it work but it does not work on boot.

vnguyen13, 2019-07-16:
@AustinPavlas That's awesome you got that one working! How did you get it to work manually?

AustinPavlas, 2019-07-16:
By running this file I am able to get the screen to run properly.  

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/apps/stats.py
This should be the same script that is initiated on boot but for some reason it only works when I manually run it.

vnguyen13, 2019-07-16:
Interesting! 

@jaybdub any thoughts on @AustinPavlas's issue on it not running the script on boot for that screen?

vnguyen13, 2019-07-16:
@AustinPavlas while we're at it.. can you share how it's wired up?

AustinPavlas, 2019-07-16:
Current wiring is as follows:  SDA -> Pin 3 (SDA), SCL -> Pin 5 (SCL), GND -> Pin 6 (GND), and VIN -> Pin 1 (3V3).  

Currently on boot the screen either remains completely blank or fills with random pixels.  
I can fix this by connecting the RST pin to GND on the Nano then running the provided stats.py scripts which produces the expected output.  I have tried running the create_stats_service.py scripts which should configure the stats.py to run a service upon boot.  I have also tried to running the command via crontab to automatically run shortly after boot, but this does not seem to work either.

jaybdub, 2019-07-16:
Odd that stats.py would run when executed standalone, but not at boot.

Can you check to make sure the system service is enabled?

```bash
sudo systemctl enable jetbot_stats
sudo systemctl start jetbot_stats
```

Also, can you dump the log into a file by calling

```bash
journalctl -u jetbot_stats >> jetbot_stats.log
```

And upload this log?  This may have some indication of what's wrong.

Best,
John

AustinPavlas, 2019-07-17:
I have double checked that the system service is enabled.  
[jetbot_stats.log](https://github.com/NVIDIA-AI-IOT/jetbot/files/3399545/jetbot_stats.log)
[image]
This is what shows on the screen after most boots, but connecting the RST pin to ground and rebooting the board fixes it most of the time.

Strangely enough, the final log about that did not show any error is was when the screen display what is shown above.

jaybdub, 2019-07-18:
Hmm,

The remote I/O error in the log is an indication that the service cannot connect with the display over I2C.

Is this possibly a hardware issue?  I notice the screen your using is a bit different.  Could you share the model as well as how you connected it to the JetBot?

Best,
John

vnguyen13, 2019-07-18:
Hi @jaybdub , @AustinPavlas mentioned that he used the following:
https://www.adafruit.com/product/931

And he gave the following information about how he connected it:

> Current wiring is as follows: SDA -> Pin 3 (SDA), SCL -> Pin 5 (SCL), GND -> Pin 6 (GND), and VIN -> Pin 1 (3V3).

DouglasPearless, 2019-07-21:
The LCD and OLED bracket uses the current chassis, it is designed to use the existing screw holes :-)

DouglasPearless, 2019-07-21:
I would expect you can hot swap the new LCD, note the OLED does not display until the JetBot has started up, it can take 1-2 mins.

vnguyen13, 2019-07-21:
Great! On your github page it doesn’t mention any specific screen. Do you have links to which one I should get?

DouglasPearless, 2019-07-22:
Ah,

I used this generic one:

https://hobbyelectronics.co.nz/product?SKU=377&Name=Mini%20128X64%20OLED%20LCD%20LED%20Display%20Module 

Cheers
Douglas

DouglasPearless, 2019-07-22:
There are plenty like this on aliexpress :-)

vnguyen13, 2019-07-24:
Thanks @DouglasPearless! How did you wire the display?

DouglasPearless, 2019-07-25:
I will post a wiring diagram tomorrow on my GitHub site :-)

vnguyen13, 2019-07-25:
Great! I ordered one in so I will be able to try it out tonight. Just let me know which wire should go where and I'll test it out. Thank you @DouglasPearless!

DouglasPearless, 2019-08-02:
Sorry, I have been delayed as I need to print out an updated chassis and rewire it all.  I now have a new chassis and can wire it, then I will post all of that on my Github account :-)

Cheers
Douglas

Cblakslee, 2019-08-29:
> By running this file I am able to get the screen to run properly.
> 
> https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/apps/stats.py
> This should be the same script that is initiated on boot but for some reason it only works when I manually run it.

I purchased the same OLED screen by mistake because the original was sold out. But mine remains pixelated no matter what. My wiring of the screen is the same as yours, 

SDA -> Pin 3 (SDA), SCL -> Pin 5 (SCL), GND -> Pin 6 (GND), and VIN -> Pin 1 (3V3).

you said connecting the RST pin to GND, then running stats.py brought up the IP address. I run the command python3 stats.py and the terminal freezes. Please let me know if I am missing something. Thank you!

AustinPavlas, 2019-08-29:
@Cblakslee 
I was never able to successfully get the OLED screen the work 100%.  I was able to get the screen to work flawlessly with another Nano that I had which was not using the Jetbot image.  Using a signal analyzer I was unable to find any differences between the two though.  In the end I was able to purchase the correct screen at a local electronics store and avoid any further headache.

Cblakslee, 2019-08-29:
> I was never able to successfully get the OLED screen the work 100%. I was able to get the screen to work flawlessly with another Nano that I had which was not using the Jetbot image. Using a signal [...]
@AustinPavlas 
Ah, ok. That's what I was hoping not to resort too, but will for now for the same reason. Thank you for the response!

closed 2022-07-15 by jaybdub