NVIDIA-AI-IOT/jetbot #107: how to display an image

seba-eng, 2019-07-16:
Hi, I'm currently struggling to display an cv2 image in the web interface. 
I wrote a very small test code to open a cv2 window, which works fine on my windows machine:
```
import cv2
import numpy as np
img = np.zeros((100, 100, 3))
cv2.imshow("Img", img)
cv2.waitKey(1000)
```

When I just run this on the jetson I get the error `gtk warning cannot open display`. 
With `export DISPLAY=:0` before that, nothing happens. The program just terminates and doesn't show anything. Same if I put it in a jupyter notebook. 
So what is the right way to display something?

jaybdub, 2019-07-16:
Hi seba-eng. 

The opencv ``imshow`` function will display using the operating systems native GUI, so it will not transfer the data over the Jupyter Web interface.  

You can however display this data using ipython widgets within a Jupyter Notebook as we do in the samples.

```python
import cv2
import numpy as np
import ipywidgets
from IPython.display import display

# create dummy image
image = np.zeros((100, 100, 3), dtype=np.uint8)

# create and display image widget
image_widget = ipywidgets.Image(format='jpeg')
display(image_widget)

# set value of image widget (must compress to JPEG to send over web).  When this value is set, the data is transferred to desktop machine and image is rendered
image_widget.value = bytes(cv2.imencode('.jpg', image)[1])
```

Please let me know if this works for you, or if you have any other questions.

Best,
John

seba-eng, 2019-07-17:
thanks, that works perfectly.

closed 2019-07-17 by seba-eng