NVIDIA-AI-IOT/jetbot #109: Slow training in jupyter notebook on Jetson nano

gprashmi, 2019-07-22:
I’m running a PyTorch model in jupyter notebook on Jetson nano and it has been very slow. The actual training doesn’t even seem to start and everything just freezes when I run the training cell. Can someone please suggest a solution for this?

jaybdub, 2019-08-10:
Hi gprashmi,

Thanks for reaching out, and sorry for the late response.  Are you still facing this issue?

Usually, this might be a sign that the process is running out of memory.  Are you able to share the script / notebook you're using to train the model?

Thanks!
John

gprashmi, 2019-08-10:
Hi John,

I am still facing the issue. You can find my notebook at the below link. 
https://github.com/gprashmi/Dog_breed_classifier/blob/master/dog_breed_classifier-6.ipynb

I am trying to run this notebook on Nano and it freezes when I run the lines:
n_epochs = 5
train(n_epochs,net,loaders,optimizer,criterion,'saved_model/dog_model.pt')

MandyTam, 2020-03-17:
Actually, I got the same problem too. 

I tried to run a Pytorch model on JupyterLab using Jetson Nano. However, it ran very slow and the camera freezed when I executed the cell for model training.

Are there any solution for this problem?

closed 2022-07-15 by jaybdub