NVIDIA-AI-IOT/jetbot #113: Add space for an encoder

ManuelZ, 2019-07-23:
It would be useful to be able to add a couple of encoders to this robot.

At the very minimum, some space would be needed for the slotted disk and for the photo interrupter.

https://www.adafruit.com/product/3985

DouglasPearless, 2019-07-25:
This is something I am looking at for my new chassis https://github.com/DouglasPearless/jetbot <https://github.com/DouglasPearless/jetbot> but it still under development :-)

wujupi, 2019-08-22:
I made one.
[image]

closed 2022-07-15 by jaybdub

mjcidcap, 2022-10-17:
Anyone have a sense of which motors, encoders, and motor drive were used in the above picture? I know the Waveshare Jetbot AI Kit uses 37-520 Metal Encoder Motor, which has an AB phase Hall encoder, but I don't think that will fit in this model.

zmswanson, 2025-10-19:
For anyone who comes across this later: the [DFRobot FIT0450 motor with encoder](https://www.digikey.com/en/products/detail/dfrobot/FIT0450/7597205) appears to be the same modified TT-style motor shown in the image above. I've also included the [datasheet](https://github.com/user-attachments/files/22988014/FIT0450_Web.pdf) in case the link breaks.

Just note that the encoder outputs use 5 V logic, so you’ll need a level shifter (or a 5 V-tolerant microcontroller) to bring the signals safely down to 3.3 V for the Jetson Nano or other controllers.

Hope that helps!