NVIDIA-AI-IOT/jetbot #114: Camera instance restarts board

rlmenge, 2019-07-26:
Hello,

I am having an issue very similar to this initialize issue : #35 . However, I replaced the camera and it did NOT fix my issue.  

As background, I am running from the battery pack (99% charge) using Jupyter notebooks in chrome (although the issue occurs when I use the terminal on the Jetbot). I am trying to run the commands:
`from jetbot import Camera
camera = Camera.instance()`

At first, running these commands would freeze the board and then trigger a full restart. I then ran `sudo systemctl restart nvargus-daemon` and now the code hangs at the camera commands.

I also included the jupyter and nvargus logs when I run the commands in the teleoperations notebook.

Thanks for any help with this!

[jetbot_jupyter_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3436083/jetbot_jupyter_log.txt)
[nvargus_daemon_log.txt](https://github.com/NVIDIA-AI-IOT/jetbot/files/3436084/nvargus_daemon_log.txt)

mike8724, 2019-10-24:
I am also having this issue, Did anyone get back to you?

jlutzwpi, 2021-05-19:
I'm not seeing a resolution to this, just links to similar issues.

1. I did the daemon restart mentioned above.
2. I tested the camera directly from the nano and it works (running gst-launch-1.0 nvarguscamerasrc ! nvoverlaysink)
3. I've tried on different browsers (Chrome and Edge) and same thing happens where the command hangs and then restarts the nano.
4. People have said that they've swapped out cameras and it works, but why would it work directly from the nano and not through jupyterlb?

Any suggestions would be appreciated.

closed 2022-07-18 by jaybdub

AICodingBot, 2024-02-29:
I'm experiencing this as well, but this code works fine on my other nano. Has anyone fixed this