NVIDIA-AI-IOT/jetbot #116: Can I not use PiOLED?

zzuzayu, 2019-08-06:
I did a good job of network setup in the early days through the network cable and HDMI screen.
I don't want to use PiOLED and I don't have PiOLED.
I closed the graphical interface.
But my terminal always shows "tegra-i2c 7000c400.i2c: no acknowledge from address 0x3c".
I want to close this tip and I need to do it.

jaybdub, 2019-08-10:
Hi zzuzayu,

Thanks for reaching out!  You can disable the display service by calling

```bash
sudo systemctl stop jetbot_stats
sudo systemctl disable jetbot_stats
```

Please let me know if this works for you, or if you have any other questions.

Best,
John

josh-github1, 2021-03-11:
The issue is sitll showing up for me and it says jetbot_stats.service is not loaded and jetbot_stats.service does not exist when I try to run both commands.

closed 2022-07-18 by jaybdub