NVIDIA-AI-IOT/jetbot #120: Collision Avoidance: when I execute "camera.observe",the car shuts down directly.

xiaodongxi2, 2019-08-19:
I saw the same problem someone posted before, his solution is set up the fps. I checked the camera file and modified this code :fps = traitlets.Integer(default_value=15).tag(config=True).However it did not work. Thus, what is the correct way to set up fps?

Mundus131, 2019-08-20:
Hey
Now i can't send you code becouse i have another issue. However i can tell you that i set up fps in Line when you are setting up size. Something like that (hight=244, width=244, fps=5). Second thing i did was replaceing observe function with infinte loop:
While True:
update()
time.sleep(0.5)

xiaodongxi2, 2019-08-20:
> Hey
> Now i can't send you code becouse i have another issue. However i can tell you that i set up fps in Line when you are setting up size. Something like that (hight=244, width=244, fps=5). Second thing i did was replaceing observe function with infinte loop:
> While True:
> update()
> time.sleep(0.5)

thank you so much! I will try it! but one thing I couldn't understand why other people can run original code successfully....

Mundus131, 2019-08-20:
I dont know but now i have big issue. When i made and ran road sign detection my jetson turned off and now he doesn't want to turn on

andrels94, 2019-09-07:
I am having this issue too. My jetbot just turns off after a while of running it with my controller. I am not sure if it is because of the camera or some motor driver failure.

xiaodongxi2, 2019-11-06:
I found out that was power problem.

closed 2019-11-06 by xiaodongxi2