NVIDIA-AI-IOT/jetbot #121: Jetson doesn't turn on

Mundus131, 2019-08-20:
Hey. 
Few moths ago I was writing about jetson which turns off while executing program . Now i have another big issue. I tried to use jetbot for detection and recognition road sign. I collected data, train AI and run program. At the begeinnig he recognised sign great and he was doing what i want, but after few sign he just turned off and now he doesn't want to turn on. All that time i was useing 10000mah xiaomi power bank. Now i even tried to power him up from phone charger but even that didn't helped me. I tried to disconnect camera, wifi adapter and no sign of life.

tokk-nv, 2019-08-21:
Hi Mundus131,
Sorry to hear you are experiencing some difficulty with the hardware.

The only thing I can think of as a potential cause to your issue was that the J48 header might be shortened with a jumper. If that is the case, the board is configured to take power only from the DC power jack and it will not take power from the Micro USB port.
Can you check if that happens to be the case?

If not, you may want to ask the same question on the Jetson Nano forum board.
https://devtalk.nvidia.com/default/board/371/jetson-nano/
There are NVIDIA support engineers on the forum who are more familiar with this kind of issue analysis, so you may have a better chance.

Mundus131, 2019-08-21:
No, there isn't any jumper at J48 header.
I just ordered recomended micro Usb power supply, which is 5,1V and 2,4 A. If it doesn't help I will ask the same question at NVIDIA support.
Thanks

jaybdub, 2019-08-22:
Hi Mundus131,

Thanks for reaching out!

It is possible that the Phone charger isn't capable of supplying the 5V@2A that Jetson Nano requires over micro USB.  

Were you previously able to power using this phone charger, but now unable too?

Best,
John

Mundus131, 2019-08-25:
Hey John
Previously i was able to power using my power bank, but sometimes jetson turns off.
I used that new power supply and jetson turned on. When i disconnected power supply and power using power bank he also truned on. So now everything is fine.
Thanks

closed 2019-08-28 by Mundus131