NVIDIA-AI-IOT/jetbot #122: Release source files for CADs?

julioasotodv, 2019-08-21:
Hi,

I am trying to mod a little bit the chassis for the Jetbot (for instance, adding a dual-camera mount) and I was wondering if it was possible to upload to this repo the source files for the 3D models (instead of just the exported STL files).

Thank you!

chitoku, 2019-08-21:
Hi julioasotodv,
Thank you for your question.

We have STEP files and IGES files listed as well as the STL files.
Do you think you can work with them?

We use Autodesk Fusion 360, so I think it is possible to also have the Fusion 360 Archive File (*.f3d), but it would be great if you can first check if you can work with those file formats first.

tokk-nv, 2019-08-21:
Hi julioasotodv,
Thank you for your question.

We have STEP files and IGES files listed as well as the STL files.
Do you think you can work with them?

We use Autodesk Fusion 360, so I think it is possible to also have the Fusion 360 Archive File (*.f3d), but it would be great if you can first check if you can work with those file formats first.

julioasotodv, 2019-08-21:
Hi @tokk-nv,

For now I am interested in modding the camera mount to support dual cameras (one firing upwards and the other one downwards), which I did by editing the provided STL file. However, this is really hacky since STL is not the best format for editing (just as PDF is not great for editing):

[image]

That said, STEP files would work for me (as I am using FreeCAD). However, the STEP file for the camera mount is unavailable in the repo right now.

Thank you!

tokk-nv, 2019-08-22:
Hi @julioasotodv ,
Thank you for letting us know that we are missing STEP file for the camera mount piece.
I filed a pull request for adding the missing STEP and IGES files, so you can find those files in the branch.
https://github.com/NVIDIA-AI-IOT/jetbot/pull/123

julioasotodv, 2019-08-22:
That sounds grear @tokk-nv!

Thank you very much (for this and the Jetbot in general; I absolutely love it 😊)

jaybdub, 2019-08-22:
Awesome, thanks Chitoku.  I've just merged @tokk-nv 's PR for camera_mount.iges / camera_mount.step into master.

@julioasotodv Looking forward to seeing what interesting things you do with the double camera setup :)

Best,
John

julioasotodv, 2019-08-26:
@tokk-nv @jaybdub awesome! Many thanks :)

I was planning to use the upwards-firing camera to develop a Yolo-like object recognition network running at the same time as the collision detector.

We can close this PR now!

closed 2019-08-26 by julioasotodv