NVIDIA-AI-IOT/jetbot #128: JetsonNano-Jetbot-Error adafruit Motor control

hungnm2014, 2019-09-05:
Dear All,

After setup software, hardware and run basic_motion , everything ok. Motor can turn left, right, backward, forward.
In collision_avoidance, I collect data, and training model OK. When i run collision_avoidance/live_demo.ipynb notebook,
Have some error occur at block 
`
from jetbot import Robot
robot = Robot()
`
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/adafruit_bus_device/i2c_device.py", line 75, in __init__
    i2c.readfrom_into(device_address, result)
  File "/usr/local/lib/python3.6/dist-packages/busio.py", line 55, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=stop)
  File "/usr/local/lib/python3.6/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 44, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end-start)
  File "/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-0.2.3-py3.6.egg/Adafruit_PureIO/smbus.py", line 155, in read_bytes
**OSError: [Errno 121] Remote I/O error**

I search some help from Adafruit to check hardware and connection i2c , everything ok, 

henry@henry-desktop:~$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --                        
henry@henry-desktop:~$ 

**but still get errno 121

Please help !

Thanks

rptallman, 2019-12-18:
I am getting the exact same error as you. However, I believe there should be an address number in hexadecimal like I do when I run the command like so:
```
jetbot@jetbot:~$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- 3d -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
The [man pages ](https://linux.die.net/man/8/i2cdetect) explain the output.

Then if I unplug the connection to the pHat and run it again:
```
jetbot@jetbot:~$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

I have not found a solution.

TCIII, 2020-03-28:
@rptallman,

If you are using the recommended DC Motor + Stepper FeatherWing Add-on the Jetbot code is expecting to see the Add-on at I2C address 0x60.

Regards,
TCIII

closed 2022-07-15 by jaybdub