NVIDIA-AI-IOT/jetbot #131: Jetbot ISO Problems: It's too slow and I can't use apt

orkunasa, 2019-09-11:
Hello,

The offical Jetson Nano ISO is working fine. Fast and rebust like any other Ubuntu version. However, Jetbot ISO is working very slowly and I can't upgrade the operating system.

Error when I try to upgrade or install anything?
[image]

Why these processes are running?
[image]

helghastGIT, 2019-09-11:
hello.
I have the same problem as you.I hope I can help you, but my English is not good, I hope you can understand.
The problem is that "apt" is said to be working. I tried to kill the process associated with "apt", but it didn't work.
I can't help you with that.
But one of my experiences you can see: I am Chinese, and in China, the official website of "Ubuntu" is very slow,  so we are used to changing sources (the file named "source.list").We are changing to China. You may reboot first, then check the link speed in your area, change the source(in your country), check in "software and update", and then "sudo apt-get update" first, and then "sudo apt-get upgrade".At least I didn't have any more problems.
I hope that helps.

helghastGIT, 2019-09-11:
the same problem(from someone) like you。
[image]

我帮你翻译一下截图里的中文。
Let me translate the Chinese in the screenshot for you.

重启后什么操作也不做，安装软件包，提示依旧。
After rebooting, do nothing, install the package, and the prompt remains the same.

可以看到系统在做每日检查更新，解决的方法很简单，关闭每日更新检查：
You can see that the system is doing the daily update check. The solution is simple: turn off the daily update check:

In the Last，System Settings | Software & Updates，In the Updates page, Automatically check for Updates and select Never
In the Updates page, Automatically check for Updates and select Never

jtiscione, 2019-09-11:
Turning off the daily update check doesn't help.

orkunasa, 2019-09-25:
I solve the locked apt resource problem.
Solution:
-  I've burned the image again
- Enter password the box which appears after login.
- Upgrade the packages.

However Jetbot image still working very slowly.

jaybdub, 2019-09-25:
Hi orkunasa,

There are a couple JetBot specific processes that run at boot.

1.  Jupyter notebook server
2.  oLED stats display server

Are you using the PiOLED display?  If not can you try disabling the stats display by the following

```bash
sudo systemctl disable jetbot_stats
sudo systemctl stop jetbot_stats
```

Please let me know if this helps.  

Also, could you share the disk usage of your system?  We've noticed that under certain circumstances one of the services will create unusually large log files.  I believe we've addressed this specific issue in JetRacer/JetCard, but not yet JetBot.  So i'm curious if this could also be part of the problem.

Best,
John

closed 2022-07-15 by jaybdub