NVIDIA-AI-IOT/jetbot #132: 关于jetbot的一些小问题（A few questions about jetbot）

helghastGIT, 2019-09-11:
关于jetbot的一些小问题
A few questions about jetbot

关于jetbot教程很少，有些jetson nano教程有讲的，但是jetbot，没说，我发现了两个，在这里总结一下，同时发布在CSDN。（打不开的人等一下审核。）
There are very few jetbot tutorials, some jetson nano tutorials are mentioned, but jetbot, not mentioned, I found two, summarized here, also published in CSDN.（Those who can't open it will be reviewed later.）

https://mp.csdn.net/postedit/100752017

我写的是中文，如果需要英语的，请您自行翻译吧。
I wrote in Chinese, if you need English, please translate it yourself.

以下原文。
The following text.

本文只是继续记录jetbot探索的过程，有些小地方可能还没有帖子写，我在这里写一下，有人需要查阅的时候可以看到。

接上回，我觉得装完镜像之后占空间还是比较大的，建议还是直接上手128G。我原来想过能不能把系统安装在外置硬盘上，无奈失败了，这里贴一位大佬的文章：https://blog.csdn.net/weixin_38153883/article/details/94436978 jetson的14G镜像和jetbot 63G镜像的分区其实应该一样的，我看就是APP区不同，但是无奈初学者，我不会弄img镜像，也希望有人知道怎么弄也交流一下。

重新购入128G TF卡后，重刷系统，超过64G Etcher就会警告，没关系继续装。

然后经过之前的经验，我先不急着上小车的驱动，当做一个Ubuntu电脑来用，先做好前期准备，换源、软件什么的，不然以后在这上面编代码还是要装。

jetbot的教程还很少，不是谁都买来做车的，很多人是当做“能够跑加速计算的树莓派”买来的，所以作为jetson nano本身的教程很多，我建议各位多去看看，因为14G和63G镜像应该是只有小车的区别，前期配置是一样的。jetbot没有直接的教程，jetson nano的教程大家可能一下子就过去了也没明白。

暂时发现两个问题：

（1）换源换什么源，很多人只用一种系统（比如16、18甚至14），所以换过之后就不管了。我就是发现我换了源之后啥也更新不了，后来想到是不是PC的CPU和ARM的镜像不一样的，不然我换了清华的源，他老说找不到东西。

我去查了jetson TX系列的帖子（大家可以去看，有借鉴的意义），我这里根据我当时查询的思路发帖子：https://blog.csdn.net/TengYun_zhang/article/details/89636497

想学习的话，这里有个详细研究的帖子，不同版本，不同平台的换源网址是不一样的：

https://blog.csdn.net/zong596568821xp/article/details/90604966

如果是jetbot搜索不到的话，其实可以搜索jetson ros这样的关键字，也是有用的：

https://blog.csdn.net/FAWKE/article/details/93590036

大家看看就明白了。（我这也算是拾人牙慧吧）

（2）很多人买了之后，可能根据店家给的，或者官网的例程，先让小车跑起来。

这里有一个步骤，说的是更新软件库后，要覆盖一下：

[image]

标灰的就是那条指令。 有些店家给的教程是用jupyter lab来写指令控制小车的（我觉得原理类似远程桌面），然后进入桌面的Notebooks文件夹：

[image]

这里有官方的引导教程，而且可以直接运行指令，让小车跑起来、运行学习功能，对于检测小车是否正常很好用。

但是！！！

刚才那个指令我发现没更新软件库（起码我的是这样），为什么呢？

 ​
[image]

注意那个叫 train_model.ipynb 的文件，旧的包没有的，更新后会在 ~/jetbot/notebooks 出现notebooks，这个是新包（看大小写），刚才的指令就是覆盖过去。

我不知道为什么没用，当然官方的教程应该是严谨的，中间出了什么差错导致的，暂时不知。不过，指令不是不能用

，我建议写成这样：

rsync -r ~/jetbot/notebooks ~/Notebooks

-r：递归模式，包含目录及子目录中所有文件

问题解决。

暂时就这些。

SHU-CF, 2019-11-01:
谢谢大佬

SHU-CF, 2019-11-01:
请问能再问一个问题嘛?
就是在做遥控手柄的那个teleoperation.ipynb这个例子时
在第三个，显示Image widget窗口时，提示运行错误，could not find module，是怎么回事儿啊，应该怎么解决呢？

SHU-CF, 2019-11-01:
麻烦你了

closed 2022-07-15 by jaybdub