NVIDIA-AI-IOT/jetbot #135: I2C not working on jetbot

jasper-hue, 2019-09-17:
Hey, 

I have been trying to get the motor driver installed , PCA9665 and lately been having the ERROR 12 - IO Error. 

jasper@ubuntu:~$ sudo i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: 
10: 
20: 
30: -- -- -- -- -- -- -- -- 
40: 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 
70: 

Shows the same in sudo i2cdetect -y 1 and in sudo i2cdetect -y 2

The wiring is correct as i have tested the same sensor with same wiring on raspberry pi and they are working perfectly fine.
Tested the system, on recovery mode too and it gives the same error. 
Also i did re-flashed the OS couple of times,and tried the same on the Jetson NANO OS. In all the cases i2C detect dosent works 

Any help would be appreciated. 

Thanks

jasper-hue, 2019-09-18:
[Update:]

I can find the devices on the console using the following command : i2cdetect -r -y 
$ i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --           

But still when i go and run  the basicmotion.py is executed the file shows error 121 : IO Error.

cagriy, 2019-10-10:
A bit late I suppose but is the PiOLED in the middle as described in the h/w setup page ? Maybe a good idea to test it without it to ensure it is not messing with any of the signals or power.

Otherwise, does the PiOLED working fine without motor driver ? Just to make sure Jetson is not faulty.

Finally, you mentioned that it was lately having this error, was it working before ?

TCIII, 2020-03-28:
@jasper-hue,
If you are using the recommended Adafruit DC Motor + Stepper FeatherWing Add-on its I2C address is not 0x40 like most servo driver boards that use the PCA9685 IC, but is 0x60. Hope that this helps.
I have used the Adafruit 16-Channel 12-bit PWM/Servo Driver in place of the Adafruit DC Motor + Stepper FeatherWing Add-on and have provided instructions here in the comments section of this [project](https://www.hackster.io/gatoninja236/building-a-custom-jetbot-with-jetson-nano-8d41b6) 

Regards,
TCIII

closed 2022-07-15 by jaybdub