NVIDIA-AI-IOT/jetbot #140: Night enviroment?

nlona, 2019-10-06:
With an ir led and without IR Filter on the cam, could the jetbot drive into a night Environment?

cagriy, 2019-10-10:
Haven't tried but cannot think of a reason for it to not be able to do this. If you are getting a clear night-vision picture on the camera widget in Jupyter, (or in the saved files after your training), learning algorithm should not struggle to create a decent model. 

I wonder if monochrome IR image reduces the quality of the model ? Any opinions from anyone else ?

nlona, 2019-10-11:
Switch from the Raspberry Pi camera to a Xiaomi mi home Security camera 360 1080p streaming wirelessly with night vision capabilities.

jaybdub, 2019-11-11:
Hi All,

This sounds very cool!

It would be awesome to see an example of one of the models trained using an IR camera.  Experiment with how it behaves in different lighting conditions, etc.

If anyone attempts this I'd love to see!

Best,
John

closed 2022-07-15 by jaybdub