NVIDIA-AI-IOT/jetbot #143: Flashing SD image & Boot

rhklite, 2019-10-10:
I've tried many times to flash the [jetson nano SD card image](https://developer.nvidia.com/jetson-nano-sd-card-image-r322), however, unsuccessful. I'm out on ideas on what to do. Hoping to find some help here.

I'll describe my issue in chronological order.

1. Flashed [Sandisk 128gb](https://www.amazon.ca/gp/product/B073JYC4XM/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1) with [Etcher 1.5.57](https://www.balena.io/etcher/) for the first time. This SD card was freshly bought. I'm not sure if Etcher said successful or not after flash.
2. Connect Nano to HDMI and inserted the Sandisk SD card. Stuck on Nvidia logo.
3. looked at [this Nvidia dev forum post](https://devtalk.nvidia.com/default/topic/1052511/jetson-nano/jetson-nano-doesn-t-boot/). It seemed like people have mostly been fixing this problem with  1) new USB cable, 2) better power supply, 3) reflash image
4. Tried 2 different USB cable, and power supplies. (didn't work)
    1. Apple Charger (5.1v-2.1A), 
    2. [Battery bank with 4.8A max current output](https://www.amazon.ca/Portable-Charger-Anker-PowerCore-20100mAh/dp/B00X5RV14Y/ref=sr_1_3?keywords=anker&qid=1570729994&sr=8-3)
5. Tried reflash disk with Etcher. At this point i noticed Etcher failed to flash. the error is `Error: EIO: i/o error, write`
6. I open the SD card in Gparted, seeing the following 2 errors and a weird looking partition (shown in 3rd image). I figured this was due to the failed flash. So cleaned up the partition and formated the disk with the following procedure.
    1. In Gparted, deleted all partitions, declare new partitions and format to ext4.
    2. In gnome-disk or nautilist GUI, format the SD card again to ext4.
[image]
[image]
[image]
7. And at this point, **I figured I was stuck on Nvidia boot screen because the SD card image wasn't properly flashed** just started trying different combinations, variations. But the result is always the same. And it always leads to the same weird looking partitions in Gparted
    1. Result 1: At 58% Etcher flash progress, write drops to 0, followed by  `Error: EIO: i/o error, write`. I reach this result if i perform step 6. then reflash 
    2. Result 2: At 12% Etcher flash progress, write drops to 0, followed by  `Error: EIO: i/o error, write`. I reach this result if i did not perform step 6 before reflash. Sometimes after this step, Gparted i would see the entire SD card as unallocated in GParted, and GParted would say there is no partition Table. I then insert new partition table using `Gparted -> Device ->Created Partition Table`, then continue from there
[image]

8. The variations that I've tried.
    1. Etcher version 1.5.57,1.5.51
    2. Doing this on Windows vs Ubuntu 16.04
    3. Using zipped vs unzip SD card image
    4. New SD card, and new SD card adapter; Kingston 32GB
    5. Redownloaded the image
    6. Using SD card image [July release](https://developer.nvidia.com/jetson-nano-sd-card-image-r322) vs [Aug release](https://developer.nvidia.com/jetson-nano-sd-card-image-r3221)
    7. Different file system format: exFAT, NTFS, ext4
    8. Using `dd` on Command Line: gets error `dd: error writing '/dev/sda1': input/output error`

Any help would be really appreciated.

cagriy, 2019-10-10:
I wouldn't expect it to be related to the power supply. In the end this is not really a power hungry operation. Few ideas, although hard to tell without seeing it.

1 - New SD card adapter fine, did that come with its own new USB cable ?
2 - Might be a good idea to try a different USB port if this is available.
3 - Probably this is not the issue, but are you sure it is the SD card on /dev/sda1 ? Instead of your local hard disk :)
4 - What is your `sudo lsblk -o NAME,FSTYPE,SIZE,MOUNTPOINT,LABEL` output ?
5 - dd should work, below is how I did it on Mac, can you try this ? 

Use lsblk or relevant command to confirm correct disk name
`sudo diskutil partitionDisk /dev/disk2 1 GPT "Free Space" "%noformat%" 100%`
`/usr/bin/unzip -p ~/Downloads/jetbot_image_v0p3p2.zip | sudo /bin/dd of=/dev/rdisk2 bs=1m`
(diskutil is a Mac command, please use equivalent command for linux)
your dd error mentions /dev/sda1 however, dd uses raw device, worth trying against /dev/rsda1 if this exists. Make sure you run dd as root, at least Mac requires root for this, perhaps linux wouldn't.

If nothing works, if possible try on another computer :)

rhklite, 2019-10-11:
@cagriy Thanks for the reply.

1. As for the adapter, i meant the microSD to SD sleeve. I didn't have a 2nd SD-USB/USB-C. I'm gonna buy a 2nd SD-USB adaptor to try this on.
2. Similar reason to 1. My adapter was a USB-C, and my computer only had one USB C port.
3. I'm sure of this, since all of my other hardrives are named /dev/nvme*
4. 
with SD card plugin in
```
NAME        FSTYPE     SIZE MOUNTPOINT                                          LABEL
loop1       squashfs    89M /snap/core/7713                                     
nvme0n1                477G                                                     
├─nvme0n1p3 ext4     276.9G /mnt/storage                                        970 Pro
├─nvme0n1p1             16M                                                     
└─nvme0n1p2 ntfs       200G /mnt/08C2120BC211FE1E                               970 Pro
loop17      squashfs   8.5M /snap/offlineimap/199                               
loop8       squashfs  35.3M /snap/gtk-common-themes/1198                        
loop15      squashfs  54.5M /snap/core18/1192                                   
loop6       squashfs 134.4M /snap/mathpix-snipping-tool/40                      
loop13      squashfs  54.4M /snap/core18/1144                                   
loop4       squashfs 193.1M /snap/mailspring/374                                
loop11      squashfs  42.8M /snap/gtk-common-themes/1313                        
loop2       squashfs   8.4M /snap/offlineimap/210                               
loop0       squashfs 135.9M /snap/cloudcompare/200                              
loop9       squashfs 255.6M /snap/kde-frameworks-5-core18/30                    
loop16      squashfs 193.3M /snap/mailspring/368                                
loop7       squashfs 420.9M /snap/pycharm-professional/154                      
nvme1n1                477G                                                     
├─nvme1n1p4 ntfs       900M                                                     WinRE tools
├─nvme1n1p2            128M                                                     
├─nvme1n1p3 ntfs     107.6G /mnt/80608CC9608CC6FE                               Windows
├─nvme1n1p1 vfat       300M /boot/efi                                           SYSTEM
├─nvme1n1p8 ext4     346.4G /                                                   
└─nvme1n1p6 ntfs      21.7G                                                     BIOS_RVY
sda                  119.1G                                                     
└─sda1      ext4     119.1G /media/richard/dc21871e-9db4-434c-98b4-713f55f807eb 
loop14      squashfs 260.7M /snap/kde-frameworks-5-core18/32                    
loop5       squashfs 135.9M /snap/cloudcompare/197                              
loop12      squashfs 421.2M /snap/pycharm-professional/159                      
loop3       squashfs  89.1M /snap/core/7917                                     
loop10      squashfs 135.9M /snap/cloudcompare/198
```
without SD card plugined in
```
NAME        FSTYPE     SIZE MOUNTPOINT                       LABEL
loop1       squashfs    89M /snap/core/7713                  
nvme0n1                477G                                  
├─nvme0n1p3 ext4     276.9G /mnt/storage                     970 Pro
├─nvme0n1p1             16M                                  
└─nvme0n1p2 ntfs       200G /mnt/08C2120BC211FE1E            970 Pro
loop17      squashfs   8.5M /snap/offlineimap/199            
loop8       squashfs  35.3M /snap/gtk-common-themes/1198     
loop15      squashfs  54.5M /snap/core18/1192                
loop6       squashfs 134.4M /snap/mathpix-snipping-tool/40   
loop13      squashfs  54.4M /snap/core18/1144                
loop4       squashfs 193.1M /snap/mailspring/374             
loop11      squashfs  42.8M /snap/gtk-common-themes/1313     
loop2       squashfs   8.4M /snap/offlineimap/210            
loop0       squashfs 135.9M /snap/cloudcompare/200           
loop9       squashfs 255.6M /snap/kde-frameworks-5-core18/30 
loop16      squashfs 193.3M /snap/mailspring/368             
loop7       squashfs 420.9M /snap/pycharm-professional/154   
nvme1n1                477G                                  
├─nvme1n1p4 ntfs       900M                                  WinRE tools
├─nvme1n1p2            128M                                  
├─nvme1n1p3 ntfs     107.6G /mnt/80608CC9608CC6FE            Windows
├─nvme1n1p1 vfat       300M /boot/efi                        SYSTEM
├─nvme1n1p8 ext4     346.4G /                                
└─nvme1n1p6 ntfs      21.7G                                  BIOS_RVY
loop14      squashfs 260.7M /snap/kde-frameworks-5-core18/32 
loop5       squashfs 135.9M /snap/cloudcompare/197           
loop12      squashfs 421.2M /snap/pycharm-professional/159   
loop3       squashfs  89.1M /snap/core/7917                  
loop10      squashfs 135.9M /snap/cloudcompare/198 
```
5. i tried dd instruction from both the official setup page, and the instruction from stereo lab's page. None worked. I forgot why the version from the official setup page didn't work. But when i ran dd the second time following the stereo lab's page, i got `dd: error writing '/dev/sda1': input/output error`. And those two sets of cmd line instructions are pretty much the same.
6. Gonna try to find a new computer to try this with :(

rhklite, 2019-10-13:
resovled with new compute + new sd card adaptor

closed 2019-10-13 by rhklite