NVIDIA-AI-IOT/jetbot #149: Basic motion import issue

mike8724, 2019-10-15:
I have followed the tutorials to the letter and up until this point haven't had any issues. I have connected to JupyterLab and started programming. The first command after checking the image files are up to date is "from jetbot import Robot"
this command line creates and error
[image]
im unable to proceed past this point.

any ideas?

David890217, 2019-11-09:
I encounter the same problem too.

tokk-nv, 2019-11-10:
Hi mik8724, David890217,

Thank you for reaching out.

It looks like the cell is incorrectly set as a code cell, not the markdown cell.
You can set it back to code by selecting the cell and change it to code from the toolbar at the top.

Regards,
Chitoku

closed 2022-07-15 by jaybdub