NVIDIA-AI-IOT/jetbot #150: Timed Photo

suptimq, 2019-10-19:
Hi

I'm wondering if there is a good way to make jetbot take photos every some (like 5 or 10...) seconds without manual clicking.

jaybdub, 2019-11-11:
Hi suptimq,

Sure!  It will require a little Python coding, but the following python code may accomplish this.  It assumes there is a directory ``images`` created that you want to dump images into.

```python
import os
import time
from jetbot import Camera, bgr8_to_jpeg

interval_seconds = 5
directory = 'images'
camera = Camera()

count = 0

while True:
    file_name = os.path.join(images, 'image_%d.jpeg' % count)
    with open(file_name, 'wb') as f:
        f.write(bgr8_to_jpeg(camera.value))
    count +=  1
    time.sleep(interval_seconds)
```

You would need to interrupt the program to stop capturing.

Please let me know if this helps or you have any questions.

Best,
John

suptimq, 2019-11-13:
Hi jaybdub,

Thanks a lot!

I have my solutions using 
```python
camera.observe(take_photo_func)
```
and it works well.

But I think your solution is better.

closed 2022-07-15 by jaybdub