NVIDIA-AI-IOT/jetbot #151: Constant camera issues

mike8724, 2019-10-24:
I cannot get my camera to display. The Jetbot moves around using basic_motion but no other action is possible because the camera isn't recognised. The camera im attempting to use is a Rasberry pi v2.1 as recommended.
The camera is set  up as per tuition with the camera ribbon in the demonstrated orientation. I have tried to run:

> sudo systemctl restart nvargus-daemon 

and get no return of errors or confirmation.

my next step was to turn the one end of the camera ribbon around and run the above step. This generated an error.

When running the Teleoperation mode. I can drive the jetbot but still cant use the camera. and no error is generated and widget fails to generate.

I moved on to data collection to see if that changed anything but got the same issues.

Im now at a loss and considering abandoning the project if no one helps me.

I have read that this is an issue with others, but would love some direction.
[image]

mike8724, 2019-10-24:
@jaybdub

jaybdub, 2019-11-11:
Hi mike8724,

Thanks for reaching out, and sorry for the delayed response on this!

I've noticed that users have experienced some issues with widgets using certain browsers.  Do you experience this issue when using Chrome?

Best,
John

mike8724, 2019-11-11:
Hi john,

I've managed to get the camera to work in chrome but not in explorer.

Its to do with Ad blocking and popups.

Thanks for your response.

Michael

jaybdub, 2019-11-11:
Great to hear!  Thanks for following up.

Best,
John

jlutzwpi, 2021-05-19:
I can get my camera to work directly from the jetson nano, but when I try to run Camera.instance() from JupyterLab, the nano crashes and restarts.  I've tried multiple browswers (IE, Chrome, Edge) and tried disabling popups.  Any suggestions?  This is driving me crazy, especially since I can get the camera to work when i'm directly connected to the nano, so I know it's not a cable/camera issue.

closed 2022-07-18 by jaybdub