NVIDIA-AI-IOT/jetbot #155: How to use jetbot.ssd_tensorrt?

shalahuddinn, 2019-11-05:
Hi, 

I want to make a custom object detection project. I want to have the jetbot can detect a red square using my own custom dataset. I have created my model to detect a red box (using my own small dataset, only 30 pictures) using Tensorflow Object Detection API. Therefore I have get my frozen graph. I am using ssd_inception_v2_coco model. I also still have the model.ckpt file.

Now, I don't understand how to convert it into engine using the jetbot.ssd_tensorrt package like what the example of object detection has done. 
[image]

heartpeng, 2021-05-07:
Have you solved it? I have the same question.

heartpeng, 2021-05-26:
import tensorflow as tf
from jetbot import ssd_tensorrt as trt
checkpoint_path = tf.train.latest_checkpoint(' ')
config_path = ' '
engine_path = ' '
my_uff = trt.ssd_pipeline_to_uff(checkpoint_path,config_path)
my_engine = trt.ssd_uff_to_engine(my_uff)
with open(engine_path,'wb') as f:
    f.write(my_engine.serialize())

I finally solved this problem in this way, but it should be noted that this package must use the previous version of object detection. I used version 0.1
Hope to help you！

huangyy1234, 2022-04-21:
> [heartpeng](/heartpeng)

Hi, could you share how you installed the object detection package in your jetbot? I ran your code and received an error that says 'No module named 'object detection. Any help will be appreciated.

closed 2022-07-18 by jaybdub