NVIDIA-AI-IOT/jetbot #158: Very unstable Jetbot demos

yahoo2016, 2019-11-07:
It's probably OK to make some impressive demo videos after many power cycles/restarts and retries, but for live demos, Jetbot Obstacle avoidance and object following demos seem very unstable.
The Object following demos stuck 9 out 10 times.
The Obstacle avoidance demo seems work 50% of the time.
When the demo got stuck, the camera display and detection meter were frozen and the Jetbot kept moving forward.

There seems multiple issues with combinations of NANO and demo software:
(1). Usages for all CPUs are around 100% even after all kernels were shutdown, in 2 or 4 CPU mode (shown in System Monitor). 
(2). Memory filled quickly and caused system sluggish even with faster Samsung EVO 128GB U3 SD cards (tested with 90MB/s sequential read).
(3). Swap usage stayed at 1.2GiB even after all kernels were shutdown and Memory usage dropped to 1.9GiB (shown in System Monitor). 

Different Nanos and Jetbots were tested, different SD cards were tried, Jetbot SD image was used without modification.  Cameras were tested using gst-lunch-1.0.
Restarting nvargus-daemon did not improve stabilities for the demos.
Are there workarounds to start the demos more reliably?

yahoo2016, 2019-11-11:
Removing "Collision Avoidance" code from "Object Following" demo seems solved the stability issue.
It appears the 2 modules (Alexnet based Collision detection and SSD-Mobilenet based Object detection) caused conflict when both are using the same camera.

jaybdub, 2019-11-11:
Thanks for sharing.  It may be that the power consumption in this scenario is exceeding the capability of the battery.  Do you witness this issue when powering from stable wall power using a sufficient (>5V@2.5A) power supply?

Best,
John

yahoo2016, 2019-11-12:
The Waveshare Jetbots have 60W power supply and 3 fully charged 18650 batteries.  3 Jetbots behave the same.  The nvargus-daemon errors only happen with combined Object following and Collision avoidance.
It seems many Jetbot owners have the same nvargus-daemon issues with the Object following demo.

yahoo2016, 2019-11-13:
It seems working after removing "return" from collision detection.

closed 2019-11-13 by yahoo2016