NVIDIA-AI-IOT/jetbot #159: How to set up a cloud GUP Machine

cystudyeverday, 2019-11-08:
Hi guys,i am here following the example3,
 "Next, we'll need to upload this data to our GPU desktop or cloud machine (we refer to this as the *host*) to train the collision avoidance neural network.  We'll assume that you've set up your training\n",

I can not find the tutorial about setting a cloud machine in JetBotWiKi, can anyone help?
Thanks for helping

jaybdub, 2019-11-08:
Hi cystudyeverday,

Thanks for reaching out!  Unfortunately, we don't explicitly specify which cloud service to use.  What we intend to say is that you can use any GPU enabled machine (including a cloud instance) to train the model.  To do this, you will need to have PyTorch installed, and download the [training notebook](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/notebooks/collision_avoidance/train_model.ipynb).

That said, for this task you can actually train the model directly on the Jetson Nano itself.  This is a relatively simple way to get started.

We have some notes on how to do this in the [examples](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Examples#option-1---train-on-jetson-nano).

Please let me know if you have any questions.

Best,
John

cystudyeverday, 2019-11-10:
Oh, i got it.
Thanks a lot for your help.
I think i also find out how to train on a cloud GPU machine by using google service.
Thanks again for helping! 
Have a nice day

jaybdub, 2019-11-11:
Great to hear!

Best,
John

closed 2022-07-15 by jaybdub