NVIDIA-AI-IOT/jetbot #160: Cannot connect to jetbot from computer

HoangLouis, 2019-11-15:
I cannot connect to my Jetbot from my laptop's web browser using _https://<jetbot_ip>:8888_
The web browser says that the Jetbot refused the connection from the laptop
[image]

kiwifruit-guo, 2019-11-23:
hey, have you already solve the problem?I met the same problem as you.

HoangLouis, 2019-11-24:
Hi there! I have done with this issue, I will lead you now. The only problem that is I flash the wrong file to my sd card.

If your PioLED is not working, you're actually the same issue as me. These are 2 SD flash files I have download and just only one work right way.
[image]

Here is the flash file (right way) that include the PioLED and you can connect easily from your computer to the Jetbot https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup
[image]

So after done flash, your PioLED will work and you can connect without the error. Hope that can help you!

Best,
Hoang Louis

closed 2019-11-24 by HoangLouis

kiwifruit-guo, 2019-11-25:
Wow，thanks so much ! 🤓Could please tell me how to get the two SD flash files? I'm looking forward to your reply.
 Best wishes.

HoangLouis, 2019-11-25:
Hi there, I just sent you a link above already https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup. Get into this link and click into this, you will have the flash file. 
[image]

Best,
Hoang Louis

kiwifruit-guo, 2019-11-25:
Thanks a lot!