NVIDIA-AI-IOT/jetbot #162: /collision_avoidance/live_demo robot.stop() not working

DennisFaucher, 2019-11-19:
Thanks for the great code. I'm up to the collision avoidance phase and everything is working. Yay. I cannot get my JetBot to stop running though once it has started. Any ideas? The code in the notebook does not stop the JetBot. TIA.

camera.unobserve(update, names='value')
robot.stop()

jaybdub, 2019-11-20:
Hi DennisFaucher,

Thanks for reaching out! 

Actually, this is a small issue with the notebook.  It turns out that after calling ``camera.unobserve``, sometimes there will be 1 frame queued even after robot.stop is called (since they are in separate threads).  

A quick workaround would be to add a small delay (ie: ``time.sleep(0.5)``) after ``camera.unobserve``.

Best,
John

DennisFaucher, 2019-11-20:
@jaybdub Thank you. I'll try this. I am also going to try and add GPU % to the OLED as I know models can take a while to load on the Nano (and the Xavier) before the next step can occur. Do you know where the code for the JetBot OLED is?

jaybdub, 2019-11-20:
Cool!  The display code is in ``stats.py``.  This script runs on boot. 

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/apps/stats.py

By the way, we did this in our ``JetCard`` project.  But we haven't incorporated in JetBot yet.

In case it helps,

https://github.com/NVIDIA-AI-IOT/jetcard/blob/master/jetcard/utils.py

Best,
John

closed 2022-07-15 by jaybdub