NVIDIA-AI-IOT/jetbot #163: Runtime Error Training 1100 Images

dcoe7, 2019-11-21:
I am trying to use the train model notebook with 550 images in free and blocked, however I am now getting a runtime error I'm having trouble finding the cause too.

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-8-f318169c73a1> in <module>
     13         outputs = model(images)
     14         loss = F.cross_entropy(outputs, labels)
---> 15         loss.backward()
     16         optimizer.step()
     17 

/usr/local/lib/python3.6/dist-packages/torch/tensor.py in backward(self, gradient, retain_graph, create_graph)
    100                 products. Defaults to ``False``.
    101         """
--> 102         torch.autograd.backward(self, gradient, retain_graph, create_graph)
    103 
    104     def register_hook(self, hook):

/usr/local/lib/python3.6/dist-packages/torch/autograd/__init__.py in backward(tensors, grad_tensors, retain_graph, create_graph, grad_variables)
     88     Variable._execution_engine.run_backward(
     89         tensors, grad_tensors, retain_graph, create_graph,
---> 90         allow_unreachable=True)  # allow_unreachable flag
     91 
     92 

RuntimeError: cuda runtime error (59) : device-side assert triggered at /home/nvidia/Downloads/tmp/pytorch/aten/src/THC/generic/THCTensorMath.cu:24

dcoe7, 2019-11-22:
Was able to do 96 images, but then I did 90 images and I got a similar error:

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-9-f318169c73a1> in <module>
     13         outputs = model(images)
     14         loss = F.cross_entropy(outputs, labels)
---> 15         loss.backward()
     16         optimizer.step()
     17 

/usr/local/lib/python3.6/dist-packages/torch/tensor.py in backward(self, gradient, retain_graph, create_graph)
    100                 products. Defaults to ``False``.
    101         """
--> 102         torch.autograd.backward(self, gradient, retain_graph, create_graph)
    103 
    104     def register_hook(self, hook):

/usr/local/lib/python3.6/dist-packages/torch/autograd/__init__.py in backward(tensors, grad_tensors, retain_graph, create_graph, grad_variables)
     88     Variable._execution_engine.run_backward(
     89         tensors, grad_tensors, retain_graph, create_graph,
---> 90         allow_unreachable=True)  # allow_unreachable flag
     91 
     92 

RuntimeError: cublas runtime error : the GPU program failed to execute at /home/nvidia/Downloads/tmp/pytorch/aten/src/THC/THCBlas.cu:258

jaybdub, 2019-11-22:
Hi dcoe7,

My guess is you're facing the issue mentioned in this pull request:

https://github.com/NVIDIA-AI-IOT/jetbot/pull/117/files

Can you try applying the changes to the notebook (adding ``float(...)`` casts around loss accumulation)?

Please let me know if this works for you, or you have any questions.

Best,
John

dcoe7, 2019-11-22:
I'll give that a try right now. Thanks for the quick response!

dcoe7, 2019-11-22:
> Hi dcoe7,
>
> My guess is you're facing the issue mentioned in this pull request:
>
> https://github.com/NVIDIA-AI-IOT/jetbot/pull/117/files
>
> Can you try applying the changes to the notebook (adding [...]

These specific lines of code aren't in my notebook because I'm using the train_model for Collision Avoidance not Road Following. I'm looking right now for the equivalent lines of accumulation but haven't found them yet.

dcoe7, 2019-11-23:
Re-flashed the SD card and same error with 96 images.

greenfan, 2019-12-29:
It seems this runtime error is caused by hidden files within the dataset directory, take a look here

https://github.com/NVIDIA-AI-IOT/jetbot/issues/111

Removing all hidden files from within the dataset directory resolved the runetime error on .... ---> 15 loss.backward()

Hope that helps.

closed 2022-07-15 by jaybdub