NVIDIA-AI-IOT/jetbot #166: object_following/live_demo fails on first line

DennisFaucher, 2019-11-30:
Hello,

from jetbot import ObjectDetector
model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

fails.  ssd_mobilenet_v2_coco.engine exists and is in the object following directory.

Error:

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'

Photo of object_following directory:

[image]

DennisFaucher, 2019-11-30:
I think this is related to issue https://github.com/NVIDIA-AI-IOT/jetbot/issues/126

j-vara, 2019-12-12:
I have the same issue (AttributeError: 'NoneType' object has no attribute 'create_execution_context'), but not same error than issue #126. Any ideas?

DennisFaucher, 2019-12-12:
The solution for me was to return the SparkFun JetBot and use the WaveShare JetBot. The SparkFun image has TensorRT 5.1 which is not compatible with the COCO model compiled for TensorRT 5.0.

AK51, 2020-02-14:
I have the same problem, Is there any solution?
Note: I used the SD card image.

AK51, 2020-02-15:
Does anyone have a solution? Thx
Do I need to use an older SD card image?

santaimpersonator, 2020-03-05:
One of our customers found where a previous version of the images was being hosted. Here is a link to the [forum post](https://forum.sparkfun.com/viewtopic.php?f=88&t=51549&p=213052#p213052) with details on how to get it working. Hope this helps.

Sorry about that @DennisFaucher. I am still waiting to get help with updating that engine.

lzty634158, 2020-04-01:
我的老镜像可以在B02最新板子上跑，但是就是B02板子HDMI无输出，升级最新镜像HDMI可以显示，但是模型错误跟楼主一样？
My old mirror can run on the B02 latest board, but is the B02 board HDMI no output, upgrade the latest mirror HDMI can be displayed, but the model error is the same as the building master?

naisy, 2020-04-03:
You can download the latest ssd_mobilenet_v2_coco.engine from the following link.
https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples

JetBot SD Card Version | Model
-- | --
v0.3 | ssd_mobilenet_v2_coco.engine
v0.4 (latest) | ssd_mobilenet_v2_coco.engine

xiepan610, 2020-05-16:
@DennisFaucher Hello! Have you solved your problem? I also have the problem,and I can't find only useful hlep on the net .Could you please give me a favor? Any help would be awesomely appreciated.

santaimpersonator, 2020-10-09:
@xiepan610 There is an updated model, linked above, by @naisy.

@DennisFaucher I think this issue can be closed due to the updated engine model.

closed 2020-10-09 by DennisFaucher

niiita, 2022-02-09:
Is there a coco engine that is compatible with tensorRT 6.0.1?