NVIDIA-AI-IOT/jetbot #168: There was an error when I ran the teleoperation.ipynb file

ddwy, 2019-12-06:
**When I ran to this step, there was an error:**

from jetbot import Robot
import traitlets

robot = Robot()

left_link = traitlets.dlink((controller.axes[0], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)
right_link = traitlets.dlink((controller.axes[1], 'value'), (robot.right_motor, 'value'), transform=lambda x: -x)

**IndexError: tuple index out of range**

jnie257, 2019-12-31:
Same issue... It seems like its hit or miss. I got it to run a handful of times and even got to see video output from my camera as i drove it around on the floor but i dont know what i did to get it to run. At this point, i went back to copy and pasting the code from the tutorial and the example code will not run! (Index Error: tuple index out of range). I went to HTML5 GamePad (controllers show up and i can see the buttons/axes as i press them. index is correct as well) and tried multiple controllers and got the same results (hit or miss). As I am typing this, i decided to run it again and IT WORKS... (but i changed nothing?) Try #2 (Again, no changes to the code...) and it gives me the error but the controller lets me drive both motors? Any ideas on how to fix this?

liamhan0905, 2020-04-24:
Out of curiosity, were you using the A02 or the B01 jetson nano?

jnie257, 2020-04-24:
I believe it is 4.2 and I know my board is the A02 version

closed 2022-07-15 by jaybdub