NVIDIA-AI-IOT/jetbot #170: Importing is too slow, it takes 13 seconds.

wesnjazz, 2019-12-07:
Importing jetbot module is too slow.
It takes 13 seconds.

To test my jetbot, I always have to wait for 13 seconds every time. (running .py, not jupyter notebook)
It's so much of wasting time.
How can I import the jetbot module faster?

dmccreary, 2020-03-15:
I have similar problems with performance.  Things that should take 100 msec take 10 seconds.

jaybdub, 2020-03-15:
Hi All,

Thanks for sharing.

This is likely due to the TensorFlow dependency.  Please let me know if you also experience this when calling ``import tensorflow``.

If so, perhaps we should consider lazily importing TensorFlow since it is no longer regularily used in the project.  (It is used for exporting the Object Detection model, which is now provided pre-compiled).

Best,
John

wesnjazz, 2020-06-22:
It still takes 11 seconds with new freshly flashed SD card image(jetbot image based on Jetpack 4.3).
How can I fix this?

```
jetbot@jetson-4-3:~/Dev/Jetbot$ time python3 -c "import jetbot"

real	0m11.147s
user	0m4.468s
sys	0m0.464s

```

wesnjazz, 2020-06-22:
This is the result of importing tensorflow
I tried three times.

```
jetbot@jetson-4-3:~/Dev/Jetbot$ time python3 -c "import tensorflow"
2020-06-21 21:13:38.832482: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0

real	0m41.388s
user	0m11.940s
sys	0m1.572s
jetbot@jetson-4-3:~/Dev/Jetbot$ time python3 -c "import tensorflow"
2020-06-21 21:14:23.858175: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0

real	0m28.596s
user	0m11.456s
sys	0m1.028s
jetbot@jetson-4-3:~/Dev/Jetbot$ time python3 -c "import tensorflow"
2020-06-21 21:14:56.921242: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0

real	0m32.459s
user	0m11.752s
sys	0m0.872s

```

closed 2022-07-15 by jaybdub