NVIDIA-AI-IOT/jetbot #171: object_following not performing collision avoidance

DennisFaucher, 2019-12-21:
Hello. I am using the WaveShare JetBot and the normal collision_avoidance notebook works great. I just had to slow the speed down a bit as the WaveShare JetBot runs fast. When I stepped up to the the object_following notebook, the camera would recognize objects (a chair in my case), place a blue box around the chair, and then run in a staight line until it crashed into something. The camera image and the blue box never changed. Has anyone else had this same problem? TIA.

DennisFaucher, 2019-12-22:
I slowed the JetBot down and it seems to be acting better

speed_widget = widgets.FloatSlider(value=0.2, min=0.0, max=0.4, description='speed')

jaybdub, 2020-03-15:
Hi DennisFaucher,

Thanks for sharing your experience!  

I've noticed before that the WaveShare JetBot is much faster than the original.  A way to mitigate this may be to adjust the motor gains

```python
left_motor.alpha = 0.5
```

Then all calls to setting the ``left_motor.value`` will run at half speed.

Best,
John

DennisFaucher, 2020-03-15:
Thanks @jaybdub The WaveShare is a great platform

rahulsingh701, 2020-06-17:
@jaybdub where do i make the changes for left_motor.value my waveshare robot is also fast?

tedchina, 2020-06-29:
@DennisFaucher the other way to slow down the bot is to use a smaller input parameters to the wrapped Robot class functions, such as `robot.forward(0.1)` or `robot.left(0.1)`.

However I guess your issue may not be due to the bot's speed, rather that no new image has been fed to the execute funtion after `execute({'new': camera.value})`, you may need to run the following `camera.observe(execute, names='value')` to trigger the live streaming feed.

closed 2022-07-15 by jaybdub