NVIDIA-AI-IOT/jetbot #174: No space for a LIDAR

fantauzzi, 2020-01-05:
Consider adding a platform on top of the jetbot to support a small and inexpensive 2D LIDAR (of the kind of  RPLiDAR A1M8 and YDLIDAR-X4).

jaybdub, 2020-03-15:
Hi Fantauzzi,

Thanks for reaching out!  This is likely outside the scope of our project, but it would be really cool to see.

If you end up creating an extension, consider sharing it here or creating a fork so others can benefit!

Best,
John

liamhan0905, 2020-04-24:
I'm actually considering doing this! I have the B01 and made some hardware addition so that the antenna can be attached to the robot. Once I get to using the lidar, I'll for sure share it too!

fjp, 2021-06-08:
You can find a slightly modified version of Jetbot in the [remo_description](https://github.com/ros-mobile-robots/remo_description) ROS package. It is designed to be modular and allows using different addons (e.g. lidar platform, camera mounts and single board computers). 

[[image]](https://media.githubusercontent.com/media/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/remo-rviz-spin.gif)

Currently there is only one lidar platform for the Slamtec RPLidar A2 but it should be easy to add a new platform that is suitable for other lidar models. 

Additionally there are camera mounts for three different camera types. The main camera holder makes it easy to switch between different camera types.

| Raspicam v2 with IMX219 | OAK-1 | OAK-D |
|:-----------------------:|:-----:|:-----:|
| [[image]](https://raw.githubusercontent.com/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/camera_types/raspi-cam.png) | [[image]](https://raw.githubusercontent.com/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/camera_types/oak-1.png) | [[image]](https://raw.githubusercontent.com/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/camera_types/oak-d.png) |

And finally you can choose between two "decks" to mount either Nvidia Jetson Nano or a Rasbperry Pi. 

| Jetson Nano | Raspberry Pi 4 B |
|:-----------------------:|:-----:|
| [[image]](https://raw.githubusercontent.com/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/sbc_types/jetson-nano.png) | [[image]](https://raw.githubusercontent.com/ros-mobile-robots/ros-mobile-robots.github.io/main/docs/resources/remo/sbc_types/raspi.png) |

If you are interesed, you can 3d print all the [stl files](https://github.com/ros-mobile-robots/remo_description/tree/main/meshes/remo) and make use of the ROS packages in the [`diffbot`](https://github.com/ros-mobile-robots/diffbot) repository to drive this robot in Gazebo simulation and the real world. This is all in early stage but additional documentation of all packages can be found at https://ros-mobile-robots.com/packages/remo_description.html.

jaybdub, 2021-06-08:
@fjp Very cool!  Love the design.

Hope this will help others looking to explore these configurations :)

Best,
John

closed 2022-07-18 by jaybdub