NVIDIA-AI-IOT/jetbot #177: switch gamepad in xbox mode not working

desvelado, 2020-01-17:
Hi,

I have a waveshare jetson nano robot kit with a shan wan gamepad from china. In jupyter python notebook its written to press 7s home button to switch the gamepad into xbox mode. At pc win 10 its workjng but not in nvidia jetson nano ubuntu 18.04 lts.
Whats the problem or how to solve it? The index of the gamepad by testing at html5gamepad.com is always 0 and testing in android mode works. If i press the home button, the browser site switches to home :d that means chromium switches to google.com.
Is it the wrong driver?
How to figure out which drivers i using?
How to get to xbox mode for teleoperating jetbot?

Greetings & thanks from martin!!

tedchina, 2020-01-19:
Waveshare says Linux is not supported. I got over this by this command to install the joystick drive in Ubuntu 18.04
`sudo apt-get install xboxdrv`

desvelado, 2020-01-19:
Now i understand - the teleoperations could be done of course from every pc not inly from jetbot's nvidia-Linux .. now the telelperation works on win-pc :)

desvelado, 2020-01-19:
Thanks a lot for solution

closed 2020-01-19 by desvelado