NVIDIA-AI-IOT/jetbot #178: i can't boot !

yizhiweimengxiangfendoudefeifei, 2020-01-21:
i have tried many times ,and i do it follewed by https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup & https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit, but my nano can't boot up, the display screen doesn't work. why? i use nano usually two weeks ago, everything is ok that time.

yizhiweimengxiangfendoudefeifei, 2020-01-21:
i really somebody help me to solve the problem, i have tried manydays.

rakshitraj, 2020-01-31:
Try writting the OS once more. Wait for Etcher to finish verifying. Then try to boot again.
Should yu get any error thy to take a screenshot or apicture and post in here. 
The alternative is that your jetson nano has gone bad. That would be unfortunate. U'll probably need to check your GPIO pins and see if the serial connections are working. U could try to debug using a UART bridge.
But lets not get ahead of ourselves. Try flashing it once more and update.

rakshitraj, 2020-01-31:
Also check if the LED glows when you supply power to the nano board. Its a green LED near the USB-b 
port

yizhiweimengxiangfendoudefeifei, 2020-02-28:
i think my jetson nano has gone bad, so i posted it to merchant. Anyway, thanks a lot.

closed 2022-07-15 by jaybdub