NVIDIA-AI-IOT/jetbot #180: Nvidia Jetbot PiOLED Issues

T-cisco, 2020-01-28:
Hello,

I am currently working with a jetbot, specifically the one sold by Waveshare and I am having issues with the piOLED. I am trying to modify it show the battery percentage. Is this possible, I have managed to get the LED to update to something but then when I restarted the jetbot the LED does not turn on. Is it possible to have the battery percentage show up on this display and if so how do I go about it?Any help will be appreciated.

fantauzzi, 2020-02-14:
I believe that what you ask for is implemented in the [Waveshare repository](https://github.com/waveshare/jetbot), which is forked from this one. You can look at their source code to see how they did it.

dmccreary, 2020-03-15:
There are APIs to see the voltages in internal structures on the Nano.  However, they don't really give you any indication of the charge remaining in the battery.  For $15 you can get a 10K mAh battery with a percent remaining display:

https://www.amazon.com/GETIHU-Portable-High-Speed-Flashlight-Compatible/dp/B07CZCV8FC

closed 2022-07-15 by jaybdub

ericlee310quark, 2023-05-08:
If you buy it from Waveshare, but download the image from official Jetbot, then the docker inside the system may be modified.
You will need to copy a new docker container which contain the files in this /jetbot repository. Then, you have to copy the Waveshare repository and overwrite some files inside the folder. After that, just set this new docker container to activate automatically after booting the machine. Finally, you will see the battery percentage display on piOLED after rebooting.