NVIDIA-AI-IOT/jetbot #184: Jetbot beeps while moving

ronak25, 2020-02-17:
Assembled Waveshare Jetbot as here: https://www.waveshare.com/wiki/JetBot_AI_Kit_Assemble_Manual. Everything works fine but whenever Jetbot moves, it causes beeping noise. Jetbot moves but noise seems unhealthy. Same behavior with both gamepad and basic_motion. Has anyone faced this issue?

liamhan0905, 2020-03-23:
I also get similar "noise". I don't think you have to worry. The motors used for this jetbot are very cheap. As long as the noise is coming from the motors, you'll be okay. If it breaks, you can always replace it since it's less than $5

ronak25, 2020-05-02:
Seems you are correct. I checked with manufacturers too. It seems unhealthy but it's nothing to worry about.

closed 2020-05-02 by ronak25