NVIDIA-AI-IOT/jetbot #189: Connect Motor Driver to Jetson Nano without PIOled display

kshen3778, 2020-03-20:
I'm not a hardware expert but If I don't have a Pioled display, how would the motor driver be connected to the Jetson Nano? Are there specific pins I need to use for the jumper wires?

TCIII, 2020-03-29:
@kshen3778,

I assume that you are using the recommended  Adafruit DC Motor + Stepper FeatherWing Add-on?

If so, then use the instructions in Hardware Assembly Step 11 to attach 3.3V (red), GND (brown), SDA (orange), and SCL (yellow) from the motor driver to the appropriate pins on the Nano GPIO Bus: Red will go to the Nano 3.3 vdc pin, Brown will go to the Nano GND pin, Orange will go the the Nano pin 3 and Yellow will go to the Nano pin 5.

Since you are not using the PiOLED do the following:

You can disable the display service by calling

sudo systemctl stop jetbot_stats
sudo systemctl disable jetbot_stats

Regards,
TCIII

kshen3778, 2020-03-29:
Thanks, I got it.

closed 2020-03-29 by kshen3778

srikrishna-bit, 2022-04-30:
> @kshen3778,
>
> I assume that you are using the recommended Adafruit DC Motor + Stepper FeatherWing Add-on?
>
> If so, then use the instructions in Hardware Assembly Step 11 to attach 3.3V (red), GND [...]

It has working properly

srikrishna-bit, 2022-04-30:
It's working properly because I working in project it's thows error pls reply me.