NVIDIA-AI-IOT/jetbot #192: Jetbot Teleoperation Index Error

liamhan0905, 2020-03-23:
I'm using a Xbox 360 controller and HTML5 Gamepad tester is recognizing it. However, when I run the script, I'm getting an stating that the tuple index is out of range... I have no clue how to go about fixing this. Anyone else have this issue?

[image]

[image]

TCIII, 2020-03-29:
@liamhan0905,

I am using a XBox360 wired controller and the index is "0" like yours.

Did you do this "controller = widgets.Controller(index=1)  # replace with index of your controller" ?

Regards,
TCIII

liamhan0905, 2020-03-29:
Yes,

I'm getting the errors below.. I'm getting many other errors on other projects too (object avoidance etc)

So, I'm thinking of reformatting my SD card and starting from scratch. I was wondering if "Create SD Card Image From Scratch" can be skipped if I just follow the Software Setup. Does the software setup have all dependant packages all installed (e.g. traitlets)

[image]

[image]

jaybdub, 2020-07-10:
Hi liamhan0905,

Thanks for reaching out.

As @TCIII mentioned, you'll want to set the index to what the gamepad tester shows.  In your case this appears to be ``0`` so you would modify the line of code to read:

```python
controller = widgets.Controller(index=0)
```

Please let me know if this helps or you run into issues.

Best,
John

closed 2022-07-15 by jaybdub