NVIDIA-AI-IOT/jetbot #193: JetBot Train Model Error: Num_Samples should be a positive integer value, but got num_samples = 0

liamhan0905, 2020-03-23:
I'm getting Num_Samples should be a positive integer value, but got num_samples = 0 error. I recorded a couple samples of the images in the data_collection.ipynb. And since I am doing this on the host, I didn't run the `"!unzip -q dataset.zip" ` line[6] because I already had a dataset dir from the data_collection.ipynb. Did anyone have a similar issue?

[image]

tokk-nv, 2020-05-16:
Hi liamhan0905,
Sorry for the delayed response.

If you are running the training notebook on the host PC, you need to zip all the images you collected on your JetBot using the last command in the data collection notebook, then manually transfer the zip file to your host PC to run the unzip command.

liamhan0905, 2020-05-16:
When you refer the host PC, are you referring to the jetson nano on the jetbot? I was accessing the jupyterlab from my desktop. How do I transfer the zip file from my desktop to the jetson nano?

9dk8, 2020-06-28:
I am having the exact same problem, trying to train the data_set on the Jetbot ie the Jetson Nano itself. Followed the instructions but don't know why it shows ValueError: num_samples should be a positive integer value, but got num_samples=0. Appreciate if anyone could help.
[image]

tomMEM, 2020-07-02:
Looks like the developer have an jetson nano independent PC installation, where they can train using the images collected by the jetson nano video camera and data_collection script. In this case, the images from the dataset directory of the jetson nano needs to be transferred to the independent training PC.

If no training PC is available, then the training is performed on the jetson nano. The images need to stay in the folder dataset with the corresponding sub-folders. Thus, the last row in the data_collection script is not necessary to activate. 
For a new data-collection session on the jetson nano, especially with a new camera or new installation, make sure that the dataset sub-folders are empty. If the camera and installation environment remains stable, then the image collection can be increased from time to time and used for additional training.

Make sure that the jeston nano is connected to the internet when the first training session is utilized - it will download one dataset that is required for the training. T

jaybdub, 2020-07-10:
Hi All,

Thanks for reaching out.

Do you have >50 images in the dataset?  The notebook manually sets a test set of 50 images, and the training set gets the remainder.

This is perhaps a bit of a flaw in the notebook.  You can try splitting the datasets by a fraction of total number instead by replacing the appropriate cell as follows

```python
train_percent = 0.9
train_count = int(train_percent * len(dataset))

train_dataset, test_dataset = torch.utils.data.random_split(dataset, [train_count, len(dataset) - train_count])
```

Please let me know if this helps or you run into any issues.

Best,
John

fariz, 2021-02-13:
My system: using SD card version
[image]

my problem is same as above please advice
1.i checked it via python3 at console (Not jupyter) with result:
   > torch is not installed
   >torchvision is not installed
  so i install it using this tutorial https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-7-0-now-available/72048
2.i restart and run again this file train_model_resnet18.ipynb
3. i still get this error 
[image]

fariz, 2021-02-13:
> Hi All,
>
> Thanks for reaching out.
>
> Do you have >50 images in the dataset? The notebook manually sets a test set of 50 images, and the training set gets the remainder.
>
> This is perhaps a bit of a [...]

if i take only 19 dataset must i change 50 with 19 ???

fariz, 2021-02-13:
help please we will get demo to student

tokk-nv, 2021-02-13:
Hi Fariz,

It looks like you did not have enough images for training.
Can you try collecting more data, like 100 images, and see if the training would go?

Another thing is that I notice that you may be running a notebook that is little outdated.
You may want to try using the latest sd card image or use the latest notebook.
https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/notebooks/collision_avoidance/train_model_resnet18.ipynb

Lastly, if you just need to demo the Collision Avoidance live, we provide the pre-trained model, as stated in the tip here.
https://jetbot.org/master/examples/collision_avoidance.html
So you may want to just grab this model and skip to the live_demo part.

fariz, 2021-02-14:
thank you very much sir, with taking more pictures in blocked and free it alread resolved
blocked = 25 picture
free.   = 25 picture

and everything resolved🙏🏻🙏🏻🙏🏻🙏🏻👍🏻👍🏻

closed 2022-07-15 by jaybdub