NVIDIA-AI-IOT/jetbot #194: I have some questions about JetBot

allen108108, 2020-03-28:
1. 
When I completed the hardware and software setup, I found that the OLED did not work when I start. I have to create another script and add a service to use it. Is this normal ?

2.
When I run the basic motion ipynb file in notebooks folder, always got the error 
`No module named ' jetbot'`. Hoe can I solve it ?

3.
Is there any tools can help me to check all setup are correct ?

TCIII, 2020-03-29:
@allen108108

1) You can use this command to see what is on the I2C bus: sudo i2cdetect -r -y 1

The command will display a grid and if the OLED is attached correctly to the Nano GPIO pins you should see 0x3c for the OLED. Also if you are using the recommended DC Motor + Stepper FeatherWing Add-on there should also see 0x60 for the Add-on. 

2) Did you perform these instructions during the software setup?

allen108108, 2020-03-30:
@TCIII 

Thanks for your answering.
I found that I flashed the wrong image to my SD card, so I reflash SD card. But something still goes wrong : 

I try to run the first example and the ipynb cells didn't get any error message,. That looks pretty good, but the engines of jetbot didn't work. That's so weird. I check all my wires of engine ,didn't wrong with them.

I don't know how to do ......

tokk-nv, 2020-04-06:
Hi allen108108,
Did you get to see anything on the OLED display?
The fact that you did not get the OLED work and failed to communicate to the motor driver suggests that the I2C wiring had some issue, so I recommend following the TCIII's advice on running the `i2cdetect` command.

closed 2022-07-15 by jaybdub