NVIDIA-AI-IOT/jetbot #195: jetson not booting with jetbot image

ndpeach, 2020-03-29:
I'm not able to boot the jetson with the jetbot SD image, no activity on the monitor. The Jetson nano powers up but there's nothing on the monitor which I've connected to the DP and the HDMI ports.

I'm able to boot the jetson nano from the normal development kit SD card image but not the jetbot images, I've tried both jetbot_image_v0p3p2 and jetbot_image_v0p4p0 and the same effect.

I've tried powering it from the 4A 5v barrel connector and via the battery / motor board in the jet bot kit, both the same result.

I've tried etcher from windows and mac with no difference, it's a 64 GB samsung SD card.

Any advice apprecaited.

TCIII, 2020-03-29:
@ndpeach,

Did you complete this instruction in the software setup:

ndpeach, 2020-03-29:
hi, thanks for your reply, yes, though in my case, I'm using the 5v 4amp barrel connector for power instead of the micro USB. Would that make a difference?  I'm following these https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup and got as far as step 2

TCIII, 2020-03-29:
@ndpeach,

I assume that you have jumper 48 installed to use the barrel power jack?

This is strange as I have had no problems using  jetbot_image_v0p4p0.zip. Did you verify the down load before you unzipped it? I always test before unzipping then use Etcher to burn the image to the SD card (San Disk 64GB Ultra Plus)

Regards,
TCIII

ndpeach, 2020-03-30:
> @ndpeach,
>
> I assume that you have jumper 48 installed to use the barrel power jack?
>
> This is strange as I have had no problems using jetbot_image_v0p4p0.zip. Did you verify the down load before you [...]

Hi, thanks again for your reply, yes, the jumper is on, I've not unzipped this and find that I can't unzip it, it looks like it's the download that is the problem, thanks for the input.

ndpeach, 2020-03-30:
hi, I've attempted to download this several times, certainly, it's causing archiver on mac and 7zip on windows to throw up errors about the zip file being corrupt, is this to do with it being on google drive, is it available somewhere else?

ndpeach, 2020-03-30:
I've now tried with the command line utilities for both mac and linux and I get the same message about a corrupt zip. here's the command and it's output, I'm just trying to see if the file will unzip.

➜  Downloads /usr/bin/unzip -p jetbot_image_v0p4p0_2.zip
[jetbot_image_v0p4p0_2.zip]
error: End-of-centdir-64 signature not where expected (prepended bytes?)
  (attempting to process anyway)
warning [jetbot_image_v0p4p0_2.zip]:  1055073 extra bytes at beginning or within zipfile
  (attempting to process anyway)
file #1:  bad zipfile offset (local header sig):  1055073
  (attempting to re-compensate)
file #1:  bad zipfile offset (local header sig):  1055073

is this file available anywhere other than google docs ? Thanks

ndpeach, 2020-03-30:
quick update, downloaded the latest zip via a linux machine, this time it's complete and not generating faults with unzip, I've been able to use etcher on linux to then copy the image to my SD card and all looks good.

closed 2020-03-30 by ndpeach