NVIDIA-AI-IOT/jetbot #196: JetBot Installation Question

liamhan0905, 2020-03-30:
Hello,

I'm having some issues all around example projects so I'm going to reinstall everything. I have couple questions below to get started.

1. Is "Create SD Card Image From Scratch" necessary step? Or can I just follow the Software Setup and just flash the SD card image?

2. What Jetpack image should I use? 4.2 vs 4.3?

tokk-nv, 2020-04-06:
Hi liamhan0905,

1. 
"Create SD Card Image From Scratch" is not necessary if you plan to just use the JetBot sdcard image.

2.
We recommend using the newer version of JetBot image, which is v0.4.0 and is based on JetPack 4.3 image.
If you are using the new, B01 revision of Jetson Nano Developer kit, you need to use JetPack 4.3 image, so you need to use JetBot image v0.4.0.

liamhan0905, 2020-04-15:
Ah Thank you! I've been getting errors when running the notebook so I wanted to just start from scratch.

liamhan0905, 2020-04-16:
Just to confirm, I do have a B01 revision jetson nano dev kit. Does that mean the Jetpack 4.2 image won't work? I wanted to implement the Gazebo ROS package from [here](https://github.com/dusty-nv/jetbot_ros) and the specification for that is Jetpack 4.2

abulnk, 2020-09-24:
I am using the default image that came with Sparkfun's Jetbot. It appears that image is based on Jetpack 4.3. They have  pre-installed dusty-nv's Jetbot ROS from [here.](https://github.com/dusty-nv/jetbot_ros). It seems the camera works but I am unable to use the motor an the OLED (both of them seem to point to Adafruit's components, not Sparfun's qwiic components). 

If the Adafruit's code on the SD works, I should be able to change to Sparkun's qwiic peripherals. The following commands do not work.

rosrun jetbot_ros jetbot_motors.py
rosrun jetbot_ros jetbot_oled.py

When I run the first one, I get the following error:

jetbot@jetson:~$ rosrun jetbot_ros jetbot_motors.py
Traceback (most recent call last):
  File "/home/jetbot/workspace/catkin_ws/src/jetbot_ros/scripts/jetbot_motors.py", line 5, in <module>
    from Adafruit_MotorHAT import Adafruit_MotorHAT
ImportError: No module named Adafruit_MotorHAT

Any suggestions would be helpful.

santaimpersonator, 2020-10-09:
@abulnk There are several files modified for this repository and the for the Jetbot ROS repository required to implement the hardware specific changes on the SparkFun kit.

If you wish to use the Adafruit hardware, your best option use the default image provided here and follow the Jetbot ROS installation instructions.

asoong98, 2021-01-13:
@santaimpersonator 
I'm currently using the SparkFun Kit version of the jetbot. Where could I find these modified files for the Qwiic hardware?

santaimpersonator, 2021-01-19:
SparkFun fork of this repo: https://github.com/sparkfun/jetbot

Currently, I am in the middle of merging the +400 latest commits from Nvidia... so I would use the latest release: https://github.com/sparkfun/jetbot/releases

santaimpersonator, 2021-06-16:
I came back cross this issue... if you no long are having problems please close the issue.

Also, I forgot to note that the `jetbot_ros` repo can be found here: https://github.com/sparkfun/jetbot_ros
- The OLED display code isn't updated for our hardware, since the `jetbot_stats.service` is running in the background on our image and works with our OLED display anyways.

closed 2022-07-18 by jaybdub