NVIDIA-AI-IOT/jetbot #197: jetbot image can't output HDMI on new B02 board?

lzty634158, 2020-04-01:
jetbot image can't output HDMI on new B02 board?

lingxiaiq, 2020-04-05:
I care about that too

tokk-nv, 2020-04-06:
Hi lzty634158 and linxiaiq,

B01 revision Jetson Nano requires JetPack 4.3 based JetBot image, that is v0.4.0.
https://github.com/NVIDIA-AI-IOT/jetbot/releases/tag/v0.4.0

I just made it clear on our release page, I'm sorry if that was not clear.
Will you make sure that you use this updated version of the image?

We tested this on both A02 and B01 Dev Kit, and I just did it again on B01 and it worked.

liamhan0905, 2020-04-18:
Hello, 

Will this B01-4.3 image be compatible with [ROS](https://github.com/dusty-nv/jetbot_ros)?

lzty634158, 2020-05-14:
@tokk-nv After several autopilot training sessions, it was found that there was not enough memory？

lzty634158, 2020-05-15:
只能重新烧写最新的镜像，重新配置环境解决此问题

closed 2020-05-15 by lzty634158