NVIDIA-AI-IOT/jetbot #198: collision avoidance working very slow

rahulsingh701, 2020-04-01:
Hi all,i recently purchased a waveshare jetbot and constructed it as per the instructions.I have also installed the latest jetbot image with jetpack 4.3.However it is seen that while running the live_demo.py 
code of collision avoidance,the robot responds to obstacle only when the video feed on the computer shows its blocked(through the slider) and therefore it respond 1-2 seconds late.Is there anyway to solve it?

rahulsingh701, 2020-04-01:
I mean to say there is a video lag and when the the video feed shows that it has detected an obstacle then it will avoid it which is causing latency of 1-2 seconds

rahulsingh701, 2020-04-01:
Is there any way to improve the latency or run the code offline so that it avoids obstacle?

rahulsingh701, 2020-04-02:
[NewZip.zip](https://github.com/NVIDIA-AI-IOT/jetbot/files/4419256/NewZip.zip)

rahulsingh701, 2020-04-02:
i have attached a video please have a look and help me asap

tedchina, 2020-04-04:
In my attempts, I didn't give such fast speed. Usually, I set it to 0.1 or 0.2, sometimes higher but 0.5 at most.

rahulsingh701, 2020-04-04:
@tedchina my jetbot motors only respond to 0.3 
Not below then that

rahulsingh701, 2020-04-04:
Is there anyway to improve the performance of jetbot cause I have seen that it reacts to obstacle only when the slider on the notebook says the path is blocked leading to 1-2 second delay?

rahulsingh701, 2020-04-08:
@jaybdub @tokk-nv can you guys help me please?

Chen-Youhui, 2020-04-14:
remove the code ' sleep(0.001) ' will reduce the delay time

Titan314v, 2020-06-11:
Did someone find a solution yet?

tokk-nv, 2020-06-11:
To see if this is somehow caused by the network, can you try to use USB Device Mode to communicate over USB?

1. Use {MicroUSB-B and Type-A} cable to connect your JetBot and PC
2. On your PC, open "192.168.55.1:8888" using your browser
3. Jupyter Lab starts. Run the same example.

jaybdub, 2020-07-10:
Hi All,

Sorry for the delayed response.

Perhaps it's a networking issue.  

If you disable any streaming, perhaps by removing the image widget (and anything referencing it) from the code, does the robot still react slowly?

Also, though it may not be the exact issue here, but if you're looking for a way to speed up the neural network inference significantly, check out [torch2trt](https://github.com/NVIDIA-AI-IOT/torch2trt).  For the road following model (resnet18) this will give ~3x speedup.

Please let me know if this helps or you run into any issues.

Best,
John

jiamei0218, 2020-10-12:
Hello, I just got Jetbot, And I have the same problem as you. Could you please help me? I have verified that it is not a network problem, the current situation is that it can run normally under the mode of 10W power supply, but it cannot run normally under the mode of 5W.Thank you so much

jiamei0218, 2020-10-13:
@jaybdub @rahulsingh701 Excuse me, can you help me?

closed 2022-07-15 by jaybdub