NVIDIA-AI-IOT/jetbot #199: Issues using Jetson Nano Revision B01

liamhan0905, 2020-04-18:
Has anyone successfully used Jetson Nano B01 version for this? 

1. I've had many issues on all the projects with my Jetson Nano B01 using Jetpack 4.3

2. I tried downloading Jetpack 4.2 on my B01 and my screen won't even load.

Let me know if anyone else had similar issues! Wondering if I should just purchase another Jetson Nano but with Version A02...

tokk-nv, 2020-05-16:
Hi liamhan0905,
I'm sorry for the delayed response.

Do you still have an issue with your B01 based JetBot?
If you are using the v0p4p0 image, and still have an issue, would you please describe the issues you have with your JetBot?

liamhan0905, 2020-05-16:
Hello,

No worries. I ended up just purchasing the A02. I was able to get the teleoperation to work. But haven't tried the other notebook yet. I'll try it over the weekend!

tomMEM, 2020-06-19:
I tried BO1 with camera attached to the upper slot (rear)- so far the scripts work nicely (last SD image version).  
Only two things I needed to do: 1) switch off "Hardware Acceleration when available" in the MicrosoftEdge (also useful to do in Chrome) and to separate the widgets to get individual "live video stream" and "vertical bar" - otherwise the system crashes within minutes.
The lag in response is about 1 second in case of the collision script - a bit faster without the video stream.
While the jetbot was activated by the script, activation of the WiFi using the  terminal while connected over micro USB to nano (sudo nmcli device wifi connect <ssid_name> password <password>)- 
and then closing browser with jupyter and disconnect the micro USB connection to the nano allowed the Jetbot to move freely. 
Reconnection over WiFi is then also possible.

closed 2022-07-15 by jaybdub