NVIDIA-AI-IOT/jetbot #201: Updated SD card instructions?

sloisel, 2020-04-27:
Hi,

I just received my NVidia Jetson Nano. I expect to get my JetBot in June hopefully, but meanwhile I wanted to make sure that my Nano is operating properly, maybe play with it a bit. I followed the official NVidia instructions [here](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit) and I'm happy to report that I'm typing this in a Chrome browser running on my NVidia Nano. So far so good.

Because I want to test some neural networks, even though I don't have the Nano yet, I thought I could start by installing the JetBot environment, which surely includes a lot of things I will need later, and also whatever is needed to do deep NN on the JetBot. When I looked up the JetBot wiki, I found I was supposed to install a custom image or make my own. Since the first step of "make my own" is to install the official NVidia image anyway, I decided to go that route. However, I am finding it difficult to follow through the official JetBot instructions [here](https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch).

I haven't got through the whole thing yet so surely I will find more things (I'll try to update this post when I do), but I've had a lot of problems, so I doubt this will work in the end.

1. I don't remember if "sudo apt install python3-pip python3-pil" worked, but I can tell you that "sudo pip3 install --upgrade numpy" did not. It gives a scary-looking error message, complete with stack trace.

2. I found some instructions for updating numpy on Ubuntu [here](https://askubuntu.com/questions/682825/how-to-update-to-the-latest-numpy-and-scipy-on-ubuntu-14-04lts). Apparently, you first have to use apt-get to remove numpy, then you can install and update it with pip3. However, that also doesn't work.

3. Apparently, you first need to pip3 install cython. I don't know why pip3 install numpy can't figure out that cython isn't installed, I thought that's what dependency management was all about. Makes me wonder if the pip3 installed package database is corrupted?

4. Earlier, I got further down the list and performed the "Install jupyter lab" steps. I don't remember if this step worked (sudo apt install nodejs npm) but I can tell you  I ended up with nodejs 8.x.x, which is a problem when you get to "sudo pip3 install jupyter jupyterlab" because jupyter requires nodejs 10.x.x or later? I found some instructions [here](https://askubuntu.com/questions/426750/how-can-i-update-my-nodejs-to-the-latest-version) on updating nodejs on ubuntu by installing the npm n package...

5. The command "sudo jupyter labextension install @jupyter-widgets/jupyterlab-manager" took an enormous amount of time to complete and brought my UI to a complete halt. I don't know if it was drawing more amps than it should have been and frying itself/the recommended USB power supply, but I subsequently dropped my nano to 5W as a safety measure. It was subsequent to this that I tried to see if I could *actually* update numpy, so that's what I'm waitong on right now.

6. I'm just completed sudo pip3 install numpy but this took an enormous amount of time and is printing a scary brown warning about using -H in sudo. I don't know what's happening.

7. sudo pip3 install scipy is completely broken. It can't find MKL or BLAS or LAPACK or anything else. I have a feeling that I broke everything.

Are there updated instructions on how to do this?

Thanks,

S

tokk-nv, 2020-05-16:
Hi sloisel,
Sorry for the delayed response.
And than you for pointing out the need of updating the "Create-SD-Card-Image-From-Scratch" wiki page. Let us add the note that it was for the JetPack 4.2 and we now have a script for the 4.3.

Yes, you may have already figured out, but we have an updated script for the SD card, that we used to create the v0p4p0 image based on JetPack 4.3.
https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/scripts/create-sdcard-image-from-scratch.sh

santaimpersonator, 2021-06-16:
@tokk-nv @jaybdub 

Is there an image for the 2GB board with Jetpack 4.3, which is compatible with that [script](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/scripts/create-sdcard-image-from-scratch.sh) and the pre-trained models *(see comments from issue #375, below)*? The oldest image that I could find was for Jetpack 4.4.1

[image]
*Source: [Jetson Download Center](https://developer.nvidia.com/embedded/downloads#?cl=version,-1)*

> hi, the pre-trained  model for the `object_following` example doesnt seem to work. I tried the file linked here for "JetBot SD Card Version" 0.4 https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docs/examples/object_following.md, and got the error below. That page warns that it only works with JetPack 4.3..
> 
> Any chance of a model compatible with 4.5? Or some tips on how to convert it, if thats possible? 
> 
> many thanks
> ```
> 
> AttributeErrorTraceback (most recent call last)
> <ipython-input-2-78c468b59bd4> in <module>
>       1 from jetbot import ObjectDetector
>       2 
> ----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')
> 
> /usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
>      27         load_plugins()
>      28         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
> ---> 29                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
>      30         self.preprocess_fn = preprocess_fn
>      31 
> 
> /usr/local/lib/python3.6/dist-packages/jetbot-0.4.3-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
>      57         with open(engine_path, 'rb') as f:
>      58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
> ---> 59         self.context = self.engine.create_execution_context()
>      60 
>      61         if input_names is None:
> 
> AttributeError: 'NoneType' object has no attribute 'create_execution_context'
> 
> ```
> 
> _Originally posted by @robbuckley in https://github.com/NVIDIA-AI-IOT/jetbot/issues/375#issuecomment-782611976_

> Hi @robbuckley ,
> 
> Thanks for reaching out!
> 
> Unfortunately, we haven't yet added support to the object detection model for JetPack 4.5.
> 
> This object detector consistently causes errors with new versions of software, likely due to the nuances in the post-processing.
> 
> We hope to develop a simple alternative to work past this.  
> 
> For the time being, you may be able to look at 
> 
> https://github.com/qfgaohao/pytorch-ssd
> 
> Or a similar open-source repository.  I'm happy to help how I can getting this to work for you.  I believe this project 
> 
> https://github.com/NVIDIA-AI-IOT/ros2_torch_trt/tree/master/trt_live_detector/trt_live_detector
> 
> Has success converting this model to TensorRT.
> 
> Please let me know if this helps or you run into any issues.
> 
> Best,
> John
> 
> _Originally posted by @jaybdub in https://github.com/NVIDIA-AI-IOT/jetbot/issues/375#issuecomment-803445929_

tokk-nv, 2021-06-16:
Hi santaimpersonator,

There is no JetPack 4.3 image for Jetson Nano 2GB, as the JetPack version predates the release of Jetson Nano 2GB Developer Kit.

santaimpersonator, 2021-06-16:
@tokk-nv Thanks!

closed 2022-07-18 by jaybdub