NVIDIA-AI-IOT/jetbot #204: jetbot object following can not run

hope450, 2020-05-06:
Program has been to stay in this code will not run down
model=Objectdetector('ssd_mobilenet_v2_coco.engine')

tokk-nv, 2020-05-16:
Hi hope450,

If you are using JetBot sdcard image v0.4 (v0p4p0), based on JetPack 4.3, you must use the updated ssd_mobilenet_v2_coco.engine (with TensorRT packaged in JetPack 4.3)

Please check this document.
https://github.com/NVIDIA-AI-IOT/jetbot/wiki/examples#example-4---object-following

xiepan610, 2020-05-16:
@tokk-nv  Hi! I  use the lastest ssd_mobilenet_v2_coco.engine,but it doesn't work.It still show a error,like this: AttributeError: 'NoneType' object has no attribute 'create_execution_context'. Could you help me solve this problem? Thank you very much!

closed 2022-07-15 by jaybdub