NVIDIA-AI-IOT/jetbot #208: Problem in object_following

xiepan610, 2020-05-16:
when I run the first step,
from jetbot import ObjectDetector model=ObjectDetector('ssd_mobilenet_v2_coco.engine')
it told me.
`---------------------------------------------------------------------------
FileNotFoundError           Traceback (most recent call last)
in
1 from jetbot import ObjectDetector
2
----> 3 model = ObjectDetector('/home/jetbot/Notebooks/object_following/ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/object_detection.py in init(self, engine_path, preprocess_fn)
26 load_plugins()
27 self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28 output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
29 self.preprocess_fn = preprocess_fn
30

/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/tensorrt_model.py in init(self, engine_path, input_names, output_names, final_shapes)
57 with open(engine_path, 'rb') as f:
58 self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59 self.context = self.engine.create_execution_context()
60
61 if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'`

xiepan610, 2020-05-16:
I can't find the way to solve it, who can help me please!

xiepan610, 2020-05-16:
I have downloaded the lastest ssd_mobilenet_v2_coco.engine,but it doesn't work

xiepan610, 2020-05-19:
I used a older image, then it gets well, but the result of test is not good. after that, I got a message said that Nvidia has since released a new engine model for their lastest image today. It may useful.

jaybdub, 2020-07-10:
Hi xiepan610,

Thanks for reaching out!  

Apologies for the delayed response.

Were you able to resolve this, or are you still running into issues?

Best,
John

lzty634158, 2020-10-29:
Hello, I am also experiencing this problem. I am in the latest Jetson Nano 2G, where Jetpack4.4.1 also reported this error @jaybdub

closed 2022-07-15 by jaybdub