NVIDIA-AI-IOT/jetbot #211: Not connecting to camera

liamhan0905, 2020-05-24:
Did anyone get this issue before? 

Things were find until recently and I can't connect to the camera anymore :(

[image]

tokk-nv, 2020-05-24:
Hi liamhan0905,

It is possible that the camera service hang so need to restart.
Would you try executing the following command on a terminal, restart the python kernel on the Jupyter notebook, then re-run?

```
sudo systemctl restart nvargus-daemon

liamhan0905, 2020-05-25:
I just tried this and I'm getting the following error 

[image]

jaybdub, 2020-07-10:
Hi liamhan0905,

Thanks for reaching out.

Any luck getting past this issue?  If the camera is still hung, and was working previously, it's possible that it is a hardware issue (like loose connector) as well.

Best,
John

Pranav-India, 2021-04-01:
camera.stop() solved the issue for me

closed 2022-07-15 by jaybdub